theyosh/TerrariumPI #429: [BUG]: AM2302 Humidity sensor 'null' / Temperature sensor works fine.

danlawless, 2020-07-13:
I have just done a clean install of Raspbian on a Raspberry Pi 3 Model B Rev 1.2 running TerrariumPI 3.9.8

[image]

I am having trouble getting a AM2302 sensor to return humidity properly from the dashboard.  I have got the temperature returning properly and see it registering.  

I have tested it in the terminal and it returns the temperature as well as the humidity.  

`root@raspberrypi:/home/pi/TerrariumPI/Adafruit_Python_DHT/examples# python3 AdafruitDHT.py 2302 4`
`Temp=28.2*  Humidity=62.6%`

Attached is a screenshot of the config.  

[image]

theyosh, 2020-07-15:
Hmm, not sure, but your address values are not correct. You should enter the physical pin number. So it should be a number between 1 and 40.

According to: https://pinout.xyz/ you are using physical pin 7 or BCM pin 4. That pin is reserved for 1-wire (https://pinout.xyz/pinout/1_wire#) . So I suggest to use a different GPIO. I use pin number 16 or 18 (BCM 23 / 24)

Then enter only the physical number of the pin as address

theyosh, 2020-08-08:
Closing due to no response....

closed 2020-08-08 by theyosh