theyosh/TerrariumPI #430: [BUG]: New installation

tetref, 2020-07-14:
I can't install the latest TerrariumPI. I followed the instructions exactly.

I have RaspberryPi 3B, latest Raspbian Buster Lite, Python 3, I also tested an older Stretch.

Only the following is displayed:

 sudo ./start.sh 3
2020-07-14 23:06:57,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server running as user 'pi' at location: http://10.0.0.26:8090 ...

And it doesn't even work as a service:

sudo_ service terriumpi restart
Failed to restart terriumpi.service: Unit terriumpi.service not found.

[image]

theyosh, 2020-07-14:
Hi, my mistake. Do a `git pull` and restart. It should work. You still get an error about no 'No GEMBIRD SiS-PM found. Check USB connections, please!' but that is just a message. It should start again.

Sorry

tetref, 2020-07-15:
Yes, it finally works, one file was changed (terrariumSwitch.py), yesterday it didn't work for git pull yet !!!

thx

blatogera, 2020-07-22:
> Hi, my mistake. Do a `git pull` and restart. It should work. You still get an error about no 'No GEMBIRD SiS-PM found. Check USB connections, please!' but that is just a message. It should start again.
> 
> Sorry

HI, I have the same error today, 

as well DHT sesors stopped responding which was good befrore the fresh install, 
also some erreor preventing apt-get update and upgrade

[image]
[image]

blatogera, 2020-07-23:
I may found the issue, however no idea how to solve it, as I just replacem my raspberry Pi 1, to Pi 4 :D 
[image]

tetref, 2020-07-23:
When I see what the problems are with DHT sensors, I switch to 1-wire and I2C.

blatogera, 2020-07-23:
> When I see what the problems are with DHT sensors, I switch to 1-wire and I2C.

Can I do that with the same sensor? Do i need any kind of extra hardware? Sorry, I'm new to this...

tetref, 2020-07-23:
I use a thermometer 18DS20 (1-wire) and I ordered BME280 and HDC1080 (I2C). No extra hw needed. And throw DHTxx in the trash ....

blatogera, 2020-07-23:
> I use a thermometer 18DS20 (1-wire) and I ordered BME280 and HDC1080 (I2C). No extra hw needed. And throw DHTxx in the trash ....

Not the solution I was expected :D but it may happen if i need to struggle a fewe more days.

theyosh, 2020-07-23:
Yeah, the owserver has changed there install a bit, and now my changes are not working.... I will write a FAQ for it to fix is.

For your DHT... throw it away :P Look at the closed issues with DHT. They work for a few days and then they start failing.

So now I use a lot of BME280 though I2C. Works nicely. You should be able to have two BME280 on 1 bus. Also some SHT2X SHT3X or HTU21D, si7021 could work with I2C. So you could use different type of hardware for humidity.

blatogera, 2020-07-23:
Thank you all for your replies! I'll replace the sensors then.
One final question if you dont mind, 

the display is showing the profile picture all the time, the system information and average readings dissapeared and i dont know how to get them back.

I will also have a read about custom display messages, on which file which section I should edit the display information?

thanks a lot again!

theyosh, 2020-07-23:
Sorry, but the display part is not really well developed or tested... Have you selected the messages you want to display. Per message you can tick where to send the message to. And the newspaper icon is display.

But again, I am currently not able to test if it is still working as should.

Custom display messages is not supported. If you want to do that, I suggest to make your own piece of code that uses my API to get the information, and make your own display setup. Then you have all the freedom you need. That is to hard to make dynamically in a web interface. So much different options and choices.

tetref, 2020-07-24:
A few more words about DHTxx .. Adafruit has stopped supporting the Adafruit_Python_DHT library and changed it to Adafruit_CircuitPython_DHT. Is it worth changing the code? DHT are terrible sensors, especially DHT22 from China ...

closed 2020-07-27 by tetref

theyosh, 2020-07-27:
Hi, I am working an a new version, which will have the new DHT library. But I still think DHT should not be used...

blatogera, 2020-07-27:
Right, changed to BME280P and HTU21D, they both works great, threw the DHT away....