theyosh/TerrariumPI #431: [BUG]: Raspberry PI Zero and DHT11

tetref, 2020-07-16:
TerrariumPI 3.9.8, RaspberryPi Zero rev.1.3 , Raspberry Pi OS Buster Lite.

I know Raspberry Pi Zero is not supported but I don't understand why DHT11 doesn't work for me.
If I run the same SD with the system on a Raspberry PI 3, then the temperature is displayed.
I found the only difference, in RPi0 I have to enter **sudo** before the command _Python3 AdafruitDHT.py 11 6_. 
For Rpi3, reading works without sudo.

In the end, I'll probably use some more modern humidity and temperature sensors anyway. 18DS20 of course works ....

theyosh, 2020-07-19:
Somehow GPIO access is only available with root rights. That is all I know.
https://github.com/theyosh/TerrariumPI/wiki/Known-issues#raspberry-pi-zero

And I am not able to fix. So either try to run it all as root, or use a different PI. Pi4 works like a charm and is pretty fast

tetref, 2020-07-20:
I had a simple project to control a terrarium with a turtle with RaspberryPi and the Domoticz system. Then I built a control with RaspberryPi Zero with TerrariumPI, where I don't need a lot of power, there won't be a camera and I have a lot of DHT11 sensors left, but ....
It looks like it doesn't have a solution and I've already ordered modern sensors.

Anyway, it's a warning to others not to use the Raspberry Pi Zero ....
[image]

theyosh, 2020-07-23:
Ok, that is clear, and a bit of a shame. But for me, it takes so much more time to develop at at PiZero, that I will not support it. A raspberry pi 3 or 4 is not that more expensive, and it works a lot faster.

But we can then close this issue?

tetref, 2020-07-24:
I think you can close this thread. I think so too. It's not about the price, but I have a finished project for Zero, including a small backup battery source, and it would be a shame to stop it due to a bad sensor.

theyosh, 2020-07-24:
Go with BME280 sensors.... do the same with I2C bus.. which should also work in a PiZero

tetref, 2020-07-24:
All sensors work, I2C and 1-wire, only those DHT with the old library do not work ....

closed 2020-07-27 by tetref