theyosh/TerrariumPI #432: [FEATURE]: Sound monitoring

havealoha, 2020-07-18:
Some components do not have an effective sensor or measurable way to detect their operation. I have started using an ultrasonic atomizer to deliver water to a root chamber for my microgreens. Im not sure if measuring the wattage used by the ultrasonic atomizer would detect if the ceramic disc has failed? One way another grower has tackled this is by monitoring sound or sound patterns generated by his misting nozzles to determine that they are in working order. This would be an interesting feature to play around with in TerrariumPI. I could see someone using it to detect bird calls, insect sounds and other audible data to control release of food or other system components or to generate an alert that something may have failed.

havealoha, 2020-07-18:
[https://aeroponicsdiy.com/high-pressure-aeroponics-spray-nozzle-monitor/](url)

theyosh, 2020-07-19:
You mean this: https://alexnld.com/product/dc-24v-45-35mm-ultrasonic-atomizer-air-humidifier/

I thinks they power usage will drop when they are not running. They should have some protection when there is not enough water. At least that says the manuals.

But sound monitoring sounds nice, but it is a bit harder to do. First question, what do I hear? Is any sound wrong? Or is any silence wrong? Do I have to recognize the sound? For me that would be hard. I have places my terrarium in my living room. So that means filtering out background noise. 

But also, this is an active process. You have to listing 24/7 else you could miss something. This is harder to make, as I cannot just ask every 30 seconds if there is sound. It could just have passed and that it is quiet again.

So I think, that sound monitoring should be something that is running outside my software.

But, you could see if it is a try to make a custom script, which will output the current sound level in dB. If that is an option, we could ad dB as a measurement value to the sensors. And that you should at least be able to see in a graph if there is sound.

But analyzing the sound to 'hear' what the sound is, is pretty hard.....

nicholseric, 2020-07-19:
Even though I might be able to detect the power drop, I am looking for a project a bit more elaborate during my Python learning. I'll see what I can do with this https://github.com/bishoph/sopare

nicholseric, 2020-07-19:
Maybe you could add triggered scripts for switches. If a directory and or file exists named "switch name" then the .py file or files in that directory would be executed. Or a field in the switch config like script enabled and path to script. That way, when my switch comes on, I could run some code that would result in a sound sample. A custom sensor would then read the result of the script?

theyosh, 2020-07-19:
Hmm, that is not handy. That could make my software unstable due to external code. I have already made some custom script sensors and switches, but I am not a fan of that. It could crash my software.
Also, your monitoring/alerting system is still depending on my software, which could lead to not being trigger at all due to some potential bugs..

I think for you, you should look in the direction of making a own small daemon that is listening 24/7 and has its own logic. And then you can make you own monitoring system. So, if you expect that you should here some sound every 15 minutes, you could trigger an email when there was no sound in 30 minutes.

For me, I think I will not add it, as it has to many variables on how people could use it.

theyosh, 2020-08-27:
For now, I do not see a need for this in my software and closing this issue. The idea is nice, but it is just way more then simple listening to sound...

closed 2020-08-27 by theyosh