theyosh/TerrariumPI #433: Issues Setting Up Environment

bcombs18, 2020-07-18:
Hello, I'm having difficulty getting my environment settings setup. I have tried using timers and the weather settings for controlling the lights. Both of these seem to flicker the lights on and off at certain intervals. I am using relays to control the lights. I also am having the same issue with the temperature settings. When the alarm min is active, it just toggles the heater on and off and then waits about 25 seconds and does it again.

Here are my configurations for each of those settings.

[image]
[image]
[image]

I currently have the relays for the lights setup as a timer because this was the only way I could get the lights to stay on during the given window of time. I would much prefer to use the environment settings instead. I am also confused about what the "Hours Shift" setting is supposed to be used for.

Thanks in advance!

bcombs18, 2020-07-19:
I believe I have solved the majority of my issues. Didn't realize I shouldn't set both the high and low alarms. I think that was what may have been causing the weird behavior. However, I am still uncertain of what the Hours Shift input is used for. I set it to 1, and I think the lights stayed on for about an hour. If I want them to stay on for something like 8 hours, would I need to set that to 8?

theyosh, 2020-07-19:
Hi,

this is only about the light settings as you have only screenshots of the temperature sensors. So for the lights settings.
First you pick a mode. Timer, or weather.

Timer: you can then select the timer window in the day part. This is usually where you put your lights. The night part can be used for infrared heating light, or some kind of blue lights... But mostly will only use the day lights.
Below the start and stop time, you can specify intervals. The interval is a period of power on and then power off. For example, 15 min on, 5 min off. And that will repeat through the timer window specified by the start and end time.

The hours shift you are referring to is only available in weather modes, where the weather source is determining when it is day and when it is light. And therefore you can then not specify the day timer values, but only the power switches. But it could be that the sun rise is to early/late and with the hours shift, you can shift when the power goes on.
The same is for min and max hours. I need at least 10 hours of light for my reptile. But during the winter, we have less hours of daylight, and therefore there will be to less light for my reptile. With the min hours, software will make sure that the minimal amount of day duration is available.

theyosh, 2020-07-19:
if you use a heater for temperature control, you should use the low alarm tab only. That part will be triggered when it is getting to cold.

If you have some kind of a cooling system, you can use that in the high alarm, which will be triggered when it is getting to hot.

theyosh, 2020-07-19:
Also, hover over the fields will give a lot of information

bcombs18, 2020-07-20:
Awesome! Thank you for your responses. That makes things a bit clearer now. I think I have gotten most of my issues sorted out so I will go ahead and close this.

Thanks again!

closed 2020-07-20 by bcombs18