theyosh/TerrariumPI #434: Forced False

cyber-shifter, 2020-07-18:
I've got a fairly simple setup at the moment. One humid/temp sensor, 4 relays for switching 120 volts. I POC'd it on the bench with breadboards, had everything hooked up. Everything was working properly, the relay switched and everything. Then I moved it, installed it in its permanent location. Sudden, the relays don't work. The button for manually switching them does nothing. I also noticed in the log that it's seeing the button push to switch it on, but enforcing a 'false" state on the switch. Can you tell me what's going on here? I'm on the latest build.

theyosh, 2020-07-19:
Hi, this sucks..... Lets see if we can find the issue.

The forced false message is a bit wrong/misleading. The force option is there to make the switch toggle even when it is already in the selected state. So you could not power it on 2 times in a row. But if you just toggle on and off, it should at least toggle once.

But there are not errors in the log? And the webinterface is also changing the state visually? Then the only conclusion is that the GPIO connections are not good. GPIO power switches do not have an option to check if they are connected and/or switched when make the toggle action. That is a downside.

Also, my GPIO relays have a small led that lights up when there is power to the board. So you know at least that there is power to operate. Then when toggle a relay that should also mostly give a led.

If that is not happening, and there are not errors in the screen session or in the log I can only think of problems in the wiring...

cyber-shifter, 2020-07-19:
Yeah, I probably should have filled in those details. The board is getting power successfully as I can see the LED for it lighting up on the board. The jumpers are set to signal on high. I'm only using one channel right now, and it's attached to pin 7 on the raspberry pi. When I toggle it manually in the interface, it changes visually. And the log changes to reflect the manual switching. Everything was set up and working before, all I did was disconnect, and reconnect in place. Is there something else I can look for in the logs? What is the behavior if there is a short somewhere?

nicholseric, 2020-07-19:
I had a similar experience and found that I had to delete the switches and re-add them before they would start working again.

cyber-shifter, 2020-07-19:
I tried that just now. Now it's doing the same thing, plus more. It forces a true when I create it, and then a false like before.

2020-07-19 17:04:44,787 - INFO    - terrariumEngine      - Update done in 0.59307 seconds. Waiting for 29.40693 seconds for next update
2020-07-19 17:04:44,667 - INFO    - terrariumSwitch      - Updated gpio power switch 'Fatima' status in 0.00004 seconds
2020-07-19 17:04:44,557 - INFO    - terrariumSensor      - Updated bme280 sensor 'Test temp' temperature from 73.79F to 73.81F in 0.00019 seconds
2020-07-19 17:04:44,442 - INFO    - terrariumSensor      - Updated bme280 sensor 'test' humidity from 51.08% to 51.09% in 0.23249 seconds
2020-07-19 17:04:44,211 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2020-07-19 17:04:32,079 - INFO    - terrariumSwitch      - Changed power switch 'Fatima' of type 'gpio' at address '7' from state 'True' to state 'False' (Forced:False)
2020-07-19 17:04:29,951 - INFO    - terrariumSwitch      - Changed power switch 'Fatima' of type 'gpio' at address '7' from state 'False' to state 'True' (Forced:False)
2020-07-19 17:04:29,193 - INFO    - terrariumEnvironment - Update done in 0.00134 seconds. Waiting for 14.99866 seconds for next update
2020-07-19 17:04:14,762 - INFO    - terrariumEngine      - Update done in 0.59076 seconds. Waiting for 29.40924 seconds for next update
2020-07-19 17:04:14,642 - INFO    - terrariumSwitch      - Updated gpio power switch 'Fatima' status in 0.00004 seconds
2020-07-19 17:04:14,532 - INFO    - terrariumSensor      - Updated bme280 sensor 'Test temp' temperature from 73.78F to 73.79F in 0.00019 seconds
2020-07-19 17:04:14,417 - INFO    - terrariumSensor      - Updated bme280 sensor 'test' humidity from 51.05% to 51.08% in 0.23191 seconds
2020-07-19 17:04:14,188 - INFO    - terrariumEnvironment - Update done in 0.00105 seconds. Waiting for 14.99895 seconds for next update
2020-07-19 17:04:14,186 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2020-07-19 17:04:13,357 - INFO    - terrariumEngine      - Done updating terrariumPI switches. Found 1 switches in 0.017 seconds
2020-07-19 17:04:13,355 - INFO    - terrariumSwitch      - Loaded power switch 'Fatima' with values: power 0.0W and waterflow 0.0L/s
2020-07-19 17:04:13,343 - INFO    - terrariumSwitch      - Changed power switch 'Fatima' of type 'gpio' at address '7' from state 'None' to state 'False' (Forced:True)
2020-07-19 17:04:13,342 - INFO    - terrariumSwitch      - Initialising 'gpio' power switch object
2020-07-19 17:04:13,342 - INFO    - terrariumEngine      - Loaded 0 excluding IDs: []
2020-07-19 17:04:13,341 - INFO    - terrariumEngine      - Updating terrariumPI switches
2020-07-19 17:03:59,172 - INFO    - terrariumEnvironment - Update done in 0.00109 seconds. Waiting for 14.99891 seconds for next update
2020-07-19 17:03:44,630 - INFO    - terrariumEngine      - Update done in 0.48095 seconds. Waiting for 29.51905 seconds for next update
2020-07-19 17:03:44,511 - INFO    - terrariumSensor      - Updated bme280 sensor 'Test temp' temperature from 73.75F to 73.78F in 0.00012 seconds
2020-07-19 17:03:44,394 - INFO    - terrariumSensor      - Updated bme280 sensor 'test' humidity from 50.94% to 51.05% in 0.23107 seconds

theyosh, 2020-07-23:
The force true value is used (as far as I still known) during startup and setup. Because when a new switch is added, I am not always able to read out the state. Then I need a state that I know it is in in. So forcing to off, is a way to do it.

But again, I am afraid that there is a cabling issue.

A way of testing is making a small piece of code like: https://gpiozero.readthedocs.io/en/stable/
Here they use a LED to toggle a GPIO pin. This can also be used for toggleing your switches. Make sure you enter a BCM port number in your test script.
By doing this, you can exclude my software, and see if it then works. If it then works, there is a software issue. If it still does not work, there is a hardware issue.

theyosh, 2020-07-28:
Any luck finding the bug/issue?

cyber-shifter, 2020-07-29:
Nope. I replaced the data wire just in case the connector was messed up inside. Still having the same issue. It's not an address issue, because the button on the GUI only responds when the address is correct. So I know I have that right. I'm still getting the forced false state in the logs. I'm not sure what else to try. All I did was move the Raspberry PI to where it belonged. It's kind of frustrating because I've got a whole setup that hinges on this, and it seems like such a small and dumb thing causing this, but I can't seem to find it.

theyosh, 2020-07-29:
Hmm, I think you not totally correct. As this is a GPIO power switch, my software will always toggle. Only when you have configured a wrong pin number (that is being used for something else). But if the pin number is correct, a GPIO switch will always succeed due to the fact the PIN is set to high.

So you cannot say that it works because my interface does toggle.

So, that is way i suggest to make a small python script just to toggle the GPIO pin.  https://gpiozero.readthedocs.io/en/stable/ You can just use the LED example.

As long as there is no secondary check, you can never find the issue.

Another thing. Are you using address 7? I think that will not work due to 1-wire https://pinout.xyz/pinout/1_wire#
And I enable 1-wire during install. So, that could be the issue. If you are not using 1-wire, you can disable it by changing the file `/boot/config.txt` and put a `#` in front of the line `dtoverlay=w1-gpio` and then reboot

cyber-shifter, 2020-07-29:
You can't hear me, but I'm laughing over here. Why? Because I am using pin 7. I totally missed that! I knew it would be something like this. I'm going to change the pin after work and try it again. Thank you so much, I totally wouldn't have thought of that!

Also, sorry if I came off forceful with my last comment. Text makes everything sound bad. I really live this platform you've made, and having it is fantastic. I was just frustrated because it felt like something dumb I was missing. Something simple. Which it is! Haha! Hopefully one day I can contribute something useful to this project. 😜

theyosh, 2020-07-29:
Yeah, I know it is something a bit hard to explain in a subtle way. Also English is not my native language.
So have one!
[image]

closed 2020-07-30 by cyber-shifter