theyosh/TerrariumPI #435: [BUG]: Wrong time setting

Hemibon, 2020-07-20:
Hi there!
I have set up a new terrarium at my workplace that I control with a TerrariumPi. Because I couldn't connect my raspberryPi to my work wifi connection (because it's a secure network with userID, password etc), I brought an unused router that I had at home to connect the PI to it. The router is not connected to the internet. 
Now, my Pi thinks it is 4hours behind the real time. Is that because the router is not connected to the internet? Is there a way to bypass this and 'tell' the Pi what its time should be? When I use my iphone as a hotspot and set up the Pi to connect to it, it works. But because I want my colleagues to be able to reach the website when I'm gone on holidays, I would like to use the router.
Thanks in advance :)
Bests,
Hemibon

Hemibon, 2020-07-20:
Btw, I'm using TerrariumPI 3.9.8 on a Pi4 Model B Rev 1.1

tetref, 2020-07-20:
I noticed that too. TerrariumPI works best with the Internet. If Raspberry does not connect to the time server, there is a different time on the display and on the website. I also thought that my timer didn't work because I set it according to the clock on the web page.

I think there is nothing left but to manually set the correct time via SSH.

theyosh, 2020-07-23:
Hi,

this is a system issue. So I cannot fix this. But I am afraid that it needs an internet connection. Also for getting online weather data, a internet connection is needed.

But I think a Raspberry PI can connect to any WiFi network. So not sure what you mean with secure network with userid and password. If it is just wpa2, you can just connect your PI.

Hemibon, 2020-07-27:
Hi, 
I finally managed to get it to work with the ethernet cable. All sorted now! I close this issue. 
Thanks for the help. 
Bests

closed 2020-07-27 by Hemibon