theyosh/TerrariumPI #437: [BUG]: webpage not loading after git pull 07232020

se7en9057, 2020-07-24:
**Setup:**
 - Python version: [3]
 - Raspberry PI: [3B+]

**Describe the bug**
After doing lates git pull which removed the support for weather YO i successfully updated my weather URL to openweathermap.org and everything seemed well. i also did a upgrade/update and restarted the Pi. now i am unable to access the webpage however the Terrarium Pi MOTD displays with temp information at SSH login.
as a test i removed the URL for weather and restarted the Pi. no change.
========================
                Falcor                              Version: 3.9.8

  Average Temperature:   97.70 F     Uptime:          21 days, 0:38:37
  Average Humidity:      25.50 %     Disk usage:      10.4 GiB (Free) / 14.3 GiB (Total)
                                    Memory usage:    643.3 MiB (Free) / 874.5 MiB (Total)
                                    CPU Load:        0.35, 0.29, 0.28
                                    CPU Temperature: 101.53 C

  Amount of power switches on: 3
  Power switch Day Light 1 is at state: on
  Power switch Heat Mat is at state: on
  Power switch Day Light 2 is at state: on

      Last update: 23-07-2020 08:09:50
==============================================================

**To Reproduce / Installation steps**
restarted Pi several times with same result.

theyosh, 2020-07-24:
Hmm, could you try to make a screenshot of the screen sesstion: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

I have two Pis running with the latest software.

se7en9057, 2020-07-24:
[image]

theyosh, 2020-07-24:
Hmm, looks like the install did not run fully. It is missing some modules.

So, rerun the installer: `./install.sh 3` this should fix it. If not, rerun the installer once more. It could be that there are some problems getting the latest libraries form online

se7en9057, 2020-07-24:
roger that. i'll give it a whirl when i get home tonight! kinda strange that the git pull then reboot caused some issues. everything had been running smoothly for a month or so. worse case i have a copy of my config so i can always just blast and try again. nuclear option.

theyosh, 2020-07-24:
If this is happening with an update.... then this is not nice! And it get the feeling that maybe the SD card is failing... time to make a backup!!! https://github.com/theyosh/TerrariumPI/wiki/FAQ#which-files-to-backup

Or did you maybe mixed up python 2 and 3?

se7en9057, 2020-07-24:
i didnt make any changes to the config. just did a git pull from the TerrarriumPI dir. everything was working for alittle bit. i updated the weather URL. then i restarted the PI shortly after that.

se7en9057, 2020-07-24:
did the ./install.sh twice no luck. i backed up settings and what not. if you dont have any additional suggestions or want any diagnostic data i'm going to do a re-install. @theyosh

se7en9057, 2020-07-25:
after doing a fresh install on a new SD card i imported my old config and was no longer able to access the web interface again. reverted the change and did a restart. web page comes back up... i'm using the old config file to essentially rebuild the server. not sure why its so mad. ive attached the backup files for analysis hopefully its just something funky with mine and not a widespread issue. 

other FYI when rebuilding the OS i used the new version of Raspberry Pi OS

[Link to files](https://drive.google.com/open?id=1R0RF04crtXO450InxVU-PfQNInI5JQCM)

theyosh, 2020-07-25:
Hm, the files are not open? I need to login at Google. And I do not have a Google account.

But this is very strange...
And the latest OS versions should be supported. I update my Pis every week with the latest OS packages... so that should not be an issue

se7en9057, 2020-07-25:
yeah i think its in the config files somewhere. i'll post the setup file below. i have a feeling its in there.

[setup.conf](https://pastebin.com/GGreCVXX)

se7en9057, 2020-07-27:
after reloading and rebuilding everything is working. if no one else is experiencing the problem i am then good. no big deal i mostly wanted to bring it to your attn just in case.

closed 2020-07-27 by se7en9057