theyosh/TerrariumPI #438: [FEATURE]: Timer Schedule

5point0, 2020-07-24:
Hi @theyosh 

Im sure this has been asked before but would it be possible to add a timer schedule?
I would like to be able to power a pump for two minutes every hour, but only every third or forth day.
So maybe something like a monthly or weekly schedule? 

Monday power on pump from 16:00-16:02, 17:00-17-02, ect.
Tuesday nothing in schedule.
Wednesday nothing in schedule.
Thursday power on pump from 16:00-16:02, 17:00-17-02, ect.

Dave0408, 2020-07-25:
Found a similar topic here: [https://github.com/theyosh/TerrariumPI/issues/231#issue-393666867](https://github.com/theyosh/TerrariumPI/issues/231#issue-393666867)

I have a comparable case. I would like to turn on my fans specifically for several short periods of the day. Independently of sensors, because I'm afraid they'd run too much otherwise.

theyosh, 2020-07-29:
@5point0 Hmm that sounds more like a calendar appointment :P But maybe the same logic can be used. Or 'abuse' the calendar for such timer settings. Because a schedule that spans more than one day is not possible at the moment. So that has to wait to the new version. Which I am working on, but goes not that fast as hoped.

@Dave0408 your timing should be able to do. Just enable the timer option at the power switch. Then you select first the total period with the fields Time start and stop fields. Then you have the fields on/of duration. Here you can then specify a pattern.

Example: every hour, 1 minute on for 10 hours a day:
Start time: 09:00, stop time: 19:00 (is the 10 hours period)
On duration: 1, off duration 59 - This means on for 1 minute, and off for 59 minutes.

And so you have a fan running for 1 minute every hour.

Dave0408, 2020-07-29:
@theyosh 
thanks, that sounds good. I guess I just misunderstood the german translation in the GUI. Or the logic behind it I had not understood. Nevertheless thank you very much.

theyosh, 2020-07-29:
The translations are not always up 2 date. I am depending on native speakers that are willing to translate. My German is not that good.

And maybe, I had chosen the wrong words in the first place...

theyosh, 2020-07-29:
And the same timer logic does also work in the Environment parts. And there you can add a sensor that acts like an extra check if it needs to power on/off

5point0, 2020-07-30:
Hi @theyosh 

Thanks for clearing this up. I know this is more of a wish!
I will wait for the new version for calendar appointments then. ;P

I have been thinking and another way to solve my dilemma would be to manually switch on the switch but then have an off timer?
So manually switch on pump and then the software would power it off after 2 minutes?
Maybe have a field in the settings of the switch to only power on for max "x" minutes unless in manual mode.

Is something like this possible?

theyosh, 2020-08-10:
Normally if this is on a daily basis, you could use the normal timers.

But if it is not on a daily basis, you could make your own small script that does the following:
 1. Put switch in manual mode
 2. Power on the switch
 3. Wait for X seconds/minutes
 4. Power off the switch
 5. Put switch back to normal mode

This can be done with the API: https://github.com/theyosh/TerrariumPI/wiki/API

Example: https://github.com/theyosh/TerrariumPI/blob/master/contrib/toggle_switch.py

theyosh, 2020-08-27:
Does the above steps help? As I am working on an new version, this feature will probably available in the new version. I am trying to make a more flexible scheduling system.

5point0, 2020-09-02:
Hi @theyosh,

I haven't played around with the yet but I probably will in the near future.
Hopefully your new version beats me to it.😅
Looking forward to it!

closed 2020-09-02 by 5point0