theyosh/TerrariumPI #440: US Alternative to Universal AC MAINS Dimmer - MPDMv4.1

Zlo98, 2020-07-26:
Hello I was wondering if anyone knows of a US alternative to the Universal AC MAINS Dimmer - MPDMv4.1. The shipping to get to me is $45 and I was hoping to find a cheaper alternative to this but if there isn’t one I will buy it. I’m not very knowledgeable on any of this stuff and don’t want to risk messing anything up by getting the wrong thing. Thanks!

theyosh, 2020-07-29:
Hi,

I know, it is expensive shipping, but I cannot find better AC dimmer. For example: https://www.instructables.com/id/Controlling-Any-Light-With-Philips-HUE/ at paragraph 8, he also mention MPDMv4.1. :+1: 

I do see a lot of AC dimmers with an Adruino. This is something that could be a soltuion. But then again I am not an electronic master, and playing with 220V is somewhat tricky.

Maybe this could be an option
https://www.amazon.com/Controller-Dimming-Regulation-Thyristor-Adjustment/dp/B07PFCVTV5/
https://www.amazon.com/IOTElectronic-Raspberry-Compatible-Controller-110V-230V/dp/B07KBV1FCS/
https://www.amazon.com/GENUINE-RobotDYN-Programmable-Controller-Compatible/dp/B071X19VL1/
https://www.ebay.com/itm/AC-LED-Bulb-Light-Dimmer-Module-Controller-Board-ARDUINO-RASPBERRY-Smart-Home/121680477317
https://www.ebay.com/itm/4CH-AC-LED-Light-Dimmer-V2-Module-Controller-Board-ARDUINO-RASPBERRY-Smart-Home/124213741826
But all those dimmers use analog input: https://electricalbaba.com/firing-angle-scr/

And for me, I cannot order all those devices to test. If you have some experience and willing to test, and get it working, I can add it.
But for now,  MPDMv4.1 is the only dimmer I found that works with PWM. Which is much easier for me to control.

theyosh, 2020-07-29:
Hmm, PWM with Adruino: https://www.instructables.com/id/AC-PWM-Dimmer-for-Arduino/ .... PWM is PWM I guess

Zlo98, 2020-07-30:
Ill check one of these out and let you know! Thank you for the help!

theyosh, 2020-08-27:
Any progress?

theyosh, 2020-11-27:
Closing due to no activity

closed 2020-11-27 by theyosh