theyosh/TerrariumPI #441: [FEATURE]: Seasonal settings

se7en9057, 2020-07-27:
**Is your feature request related to a problem? Please describe.**
Some plants require dormancy as well as animals. it would be useful if a second set of settings or environmental conditions could be set during selected calendar ranges. 
example: Carnivorous plants(some) require a period where they die back during the colder winter months. 

**Describe the solution you'd like**
To be able to set a date range to allow a second (or more) set of conditions allowing for different temp/moisture/humidity etc settings.

theyosh, 2020-08-09:
Hi sorry for the late reply. But this is a bit tricky to make. It looks a bit like #166 .

But I think that this is something you should do as a human when it is needed. I am not familiar with those plants, but I think you should not 'program' them.
So when it is time to change the season, I would do that manually. So I can make the right adjustments based on what I see. 

But, how about using the calendar for the schedule as a reminder for yourself.

Because for me I have to make a hole new part in the system when I can store all this information. Because you want to go back to a known previous state. But that state was then about 3 months ago. So then I need to store that information as well. For a human we can much easier store information than with software ;)

se7en9057, 2020-08-09:
makes sense. i know its alot to tinker with and program. still figuring out how to make things work. switches to monitor water levels. having automation operate different systems based on different scenarios. really enjoying your software platform. thanks again for all your hard work! wish there was a way for me to help contribute better.

closed 2020-08-09 by se7en9057

theyosh, 2020-08-10:
There are some handy scripts in the contrib folder which are not all mine :)

Also, you can make you own script that uses the API https://github.com/theyosh/TerrariumPI/wiki/API

This way, you could make a script that alters the sensors when it is time to hibernate....