theyosh/TerrariumPI #442: [BUG]: limit_min and limit_max not working anymore

bggangel, 2020-07-29:
Hi,

thx for all your work.
I ran into a little annoying issue today. I was playing around with some gpio sensors and added and remove some sensors. The weird thing after a lot of reboots, the graphs on my dashboard do not seem to "look at" the limit min/max values of the sensors anymore. Pretty sure this worked before.
Any idea why that happened and how I can fix this?
Attached: settings.cfg and graph 

**Setup:**
 - Python version: 3
 - Raspberry PI: 3B

**To Reproduce / Installation steps**
Not sure how it happened. I just look at the dashboard.

**Expected behavior**
The graphs y axis to look at the limit min/max values
[settings.zip](https://github.com/theyosh/TerrariumPI/files/4992468/settings.zip)
[image]

theyosh, 2020-07-29:
Hmm, it is not clear what you mean. The graphs looks good to me. The drop you have, is probably due to some changes, or working on the Pi. If that is happening only when you are changing stuff, that is normal.

If you revere to the spikes all the time, it could be that one sensor takes to much time to get the data. Maybe broken sensor.

Is see in the config you are using DHT. Drop that sensor. https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed+DHT and use a BME280

Also, you have just one temperature and one humdity sensor. But the are both the same hardware. So I guess your DHT is end of life...

bggangel, 2020-07-29:
Hi,
I know the DHT22 is problematic and the BME280 is already in the mail. That is one of the reason as I was messing with them as I had 3 DHT22's and could not believe they were that bad (and wanted to connect multiple to see what they would say).

Anyway, the question I have, while waiting for my new sensor is not related to the data or sensor but the graph itself.
I seem to remember that setting the limit min and max would limit the range of the graph.
It does not do that anymore. I am curious why that is. I can only thing that there is a "lost"/hanging setting somewhere, making it think there is multiple sensors. This sensor is not visible in the UI but perhaps somewhere in the settings? (Settings attached in previous note)
(Het bereik van de grafiek lijkt niet gelimiteerd te worden. Ik heb van 50 to 90 ingevuld, maar de grafiek toont 0 tot 100)

Dave0408, 2020-07-29:
Here it works as expected:
[image]
[image]

Last `git pull` is 4 days ago.

bggangel, 2020-07-29:
@Dave0408 : Yes, thank you. That is how it used to be with me also, before I started adding and removing sensors. I think there is a sensor setting somewhere invisible to the UI. As soon as I remove that, I am sure it will go back to normal. I just don't know where to look/change it.

theyosh, 2020-07-29:
Ah, now it is clear. No, the limit values will not control the graph. It will only control the values that are considered valid from the sensor.

So values below the low limit or above the high limit will not be counted as valid, and should be ignored. And if that is working well, your graph will stay within the limits.

But in your case, it can happen that during startup the limits are not loaded, and then the limits are 0 - 100. And you where changing some settings, it could happen that a sensor was initializing and read out the same time. And then it could give a lower value.

So, the values of the sensor only works on the readout of the sensor data. And the graph will determine the min and max values based on the measurements. Not the settings of the sensor.

bggangel, 2020-07-29:
ok, clear. Do you know why I am all seeing "Average Temperature" and not "Temperaturesensor <Name>" like I did before and in Dave0408's screenshot?

Dave0408, 2020-07-29:
@bggangel 
In the Home -> Overview section I see average values as well.
In the Sensor -> Temperatur (or Humidity) section I see every sensor itself. 

It is also possilble to setup an "All sensor" gauge tab. See [here](https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-create-a-sensor-page-overview)
Also you can exclude single sensor from beeing included in the average calculation. See [here](https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-exclude-a-sensor-from-average-calculation)

bggangel, 2020-07-29:
ok, thx! I completely remembered wrong than!
My questions are answered. Closed!
(Thx for all the great work)

closed 2020-07-29 by bggangel