theyosh/TerrariumPI #443: Trouble setting up environment setup using weather dependency

Dave0408, 2020-07-29:
@theyosh 
As you described the timer logic in to me I was able to get my fans run 2 Minutes every hour. Perfect. 
But now i want to controll my light depending on sunrise/sunset in Parintins, Amazonas, due to I've setup my terrarium with some tropical south america plants. 
First, using a timer from 08:00 - 18:00 with 600 minutes on and 0 minutes off works. 
But using the envorinment setup i had some troubles....What i observe is that light turns on and immediately off again. 
Also, sometimes not always, I'm able to edit ALL values for day/night setting. I think that makes no sense due to the global on/off setting for 10 hours?!

My setup:
[image]
[image]
[image]

Sometimes it looks like this:
[image]

My log during lights going on/off:
> 2020-07-29 15:02:34,842 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
2020-07-29 15:02:34,736 - INFO    - terrariumSwitch      - Updated mss425e power switch 'Skylight LED' status in 0.20340 seconds
2020-07-29 15:02:34,533 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
2020-07-29 15:02:34,406 - INFO    - terrariumSensor      - Updated bme280 sensor 'Temp_BME280' temperature from 22.98C to 23.01C in 0.00222 seconds
2020-07-29 15:02:34,353 - INFO    - terrariumEnvironment - Update done in 0.50592 seconds. Waiting for 14.49408 seconds for next update
2020-07-29 15:02:34,234 - INFO    - terrariumSwitch      - Changed power switch 'Skylight LED' of type 'mss425e' at address '1' from state 'True' to state 'False' (Forced:False)
2020-07-29 15:02:34,188 - INFO    - terrariumSensor      - Updated bme280 sensor 'Luftfeuchtigkeit' humidity from 71.89% to 71.14% in 0.25607 seconds
2020-07-29 15:02:34,068 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
2020-07-29 15:02:34,060 - INFO    - terrariumEnvironment - Environment light is turning off the alarm max powerswitches based on weather
2020-07-29 15:02:33,973 - INFO    - terrariumSwitch      - Changed power switch 'Skylight LED' of type 'mss425e' at address '1' from state 'False' to state 'True' (Forced:False)
2020-07-29 15:02:33,868 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
2020-07-29 15:02:33,861 - INFO    - terrariumEnvironment - Environment light is turning on the alarm min powerswitches based on weather
2020-07-29 15:02:33,841 - INFO    - terrariumAudio       - Update done in 0.00003 seconds. Waiting for 29.99997 seconds for next update
2020-07-29 15:02:33,837 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00004 seconds. Waiting for 29.99996 seconds for next update
2020-07-29 15:02:33,806 - INFO    - terrariumSensor      - Updated w1 sensor 'Temp_Boden' temperature from 21.44C to 21.44C in 0.91132 seconds
2020-07-29 15:02:32,766 - INFO    - terrariumSensor      - Updated w1 sensor 'Temp_aussen' temperature from 24.38C to 24.38C in 0.91147 seconds
2020-07-29 15:02:31,726 - INFO    - terrariumSensor      - Updated w1 sensor 'Temp_Wasser' temperature from 22.56C to 22.62C in 0.91403 seconds
2020-07-29 15:02:30,686 - INFO    - terrariumSensor      - Updated w1 sensor 'Temp_Decke' temperature from 23.81C to 23.75C in 0.84869 seconds
2020-07-29 15:02:19,260 - INFO    - terrariumEnvironment - Update done in 0.44361 seconds. Waiting for 14.55639 seconds for next update
2020-07-29 15:02:19,158 - INFO    - terrariumSwitch      - Changed power switch 'Skylight LED' of type 'mss425e' at address '1' from state 'True' to state 'False' (Forced:False)
2020-07-29 15:02:19,029 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
2020-07-29 15:02:19,022 - INFO    - terrariumEnvironment - Environment light is turning off the alarm max powerswitches based on weather
2020-07-29 15:02:18,935 - INFO    - terrariumSwitch      - Changed power switch 'Skylight LED' of type 'mss425e' at address '1' from state 'False' to state 'True' (Forced:False)
2020-07-29 15:02:18,837 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
2020-07-29 15:02:18,830 - INFO    - terrariumEnvironment - Environment light is turning on the alarm min powerswitches based on weather
2020-07-29 15:02:06,336 - INFO    - terrariumEngine      - Update done in 6.77454 seconds. Waiting for 23.22546 seconds for next update
2020-07-29 15:02:06,116 - INFO    - terrariumSwitch      - Updated gpio-inverse power switch 'Lüfter' status in 0.10107 seconds
2020-07-29 15:02:06,113 - INFO    - terrariumSwitch      - Power switch changed from True state to new False state
2020-07-29 15:02:06,018 - INFO    - terrariumSwitch      - Changed power switch 'Lüfter' of type 'gpio-inverse' at address '16' from state 'True' to state 'False' (Forced:False)
2020-07-29 15:02:05,911 - INFO    - terrariumSwitch      - Updated mss425e power switch 'USB Stecker' status in 0.21407 seconds
2020-07-29 15:02:05,698 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
2020-07-29 15:02:05,592 - INFO    - terrariumSwitch      - Updated mss425e power switch 'Pumpe' status in 0.17566 seconds
2020-07-29 15:02:05,417 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
2020-07-29 15:02:05,311 - INFO    - terrariumSwitch      - Updated mss425e power switch 'Beregnungsanlage' status in 0.23086 seconds
2020-07-29 15:02:05,081 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
2020-07-29 15:02:04,975 - INFO    - terrariumSwitch      - Updated mss425e power switch 'Skylight LED' status in 0.18884 seconds
2020-07-29 15:02:04,787 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
2020-07-29 15:02:04,657 - INFO    - terrariumSensor      - Updated bme280 sensor 'Temp_BME280' temperature from 23.02C to 22.98C in 0.25428 seconds
2020-07-29 15:02:04,376 - INFO    - terrariumEnvironment - Update done in 0.56511 seconds. Waiting for 14.43489 seconds for next update
2020-07-29 15:02:04,276 - INFO    - terrariumSwitch      - Changed power switch 'Skylight LED' of type 'mss425e' at address '1' from state 'True' to state 'False' (Forced:False)

Ps.: 
I will have a look into the wiki and see if I can help to translate to german....;-)

theyosh, 2020-07-29:
Hmm, only use Tag/Day power switches... Now the lights are on and off 24/7 Which will not work

So, you select a power switch to be on at certain times (Day). When it is outside those hours, it will be off.

The Night option is for night lights or IR heaters... So clear your night tab and it should work

Translations is pretty easy: http://weblate.theyosh.nl using your Github login.

Dave0408, 2020-07-30:
Got it....Thanks.

Dave0408, 2020-08-01:
Hello, 

i've got an other question.
Is there a way to combine a timer and an evironment setting using the same source? 
For example i've setup a humidity environment. If it reaches max threshold it will turn on my fans for 3 Minutes.  If it reaches minimum threshold it will turn on my rainingsystem for 10 seconds. That works so far. 

But i also want my fans and my rainingssystem run on a scheduled daily basis. Let's say my fans should run every morning and evening for 15 minutes and my rainingsystem should run three times a day. It doesn't matter what the state of the humidity sensor is. 
So I tried to configure a parallel timer on my fans:
[image]

What happens is that ist turn on due to the timer setting and with the next checkup cycle it will turn off by humidity setting and restart immidiately.  
```
2020-08-01 06:47:52,596 - INFO    - terrariumSwitch      - Changed power switch 'Lüfter' of type 'gpio-inverse' at address '16' from state 'False' to state 'True' (Forced:False)
2020-08-01 06:48:21,370 - INFO    - terrariumEnvironment - Environment humidity is turning off the alarm max powerswitches based on sensor
2020-08-01 06:48:21,376 - INFO    - terrariumSwitch      - Changed power switch 'Lüfter' of type 'gpio-inverse' at address '16' from state 'True' to state 'False' (Forced:False)
```
I would like to avoid such a fast on and off. for the fans it might be ok, but for other devices rather not. 
I also couldn't see any change in behaviour when making my environment depend on my light setup. (which I don't really want)

My environment setup:
[image]

Any advise? Thanks in advance.

theyosh, 2020-08-10:
Hmm, I am afraid that this will not work. It is almost the same issue as #446 https://github.com/theyosh/TerrariumPI/issues/446#issuecomment-671027790

So,when you configure a power switch with his own timer, it will not allow to be switched in a different state then the timer is instructed to. So when the environment will toggle the switch, the timer will toggle back.

So, I think you should disable the timer on the power switch, and use the timer in the humidity environment and just go for the three times spraying. And then do not depend on the sensor.
When you add the sensor, it will not guarantee that it will spray in the timer window, if the sensor is not in the alarm min state.

A solution could be to specify that a power switch can be overruled when it has a timer window. But then I have to think how that would work for other scenarios. Have to think about a bit more...

Dave0408, 2020-09-30:
Hello @theyosh 

i'm currently writing a script to control my fans.

```
#!/usr/bin/python3
import RPi.GPIO as GPIO
import time
import os
#Stop terrariumpi
os.system('systemctl stop terrariumpi.service')
#use P1 header pin numbering convention
GPIO.setmode(GPIO.BOARD)
#Set up the GPIO channels - one input and one output
GPIO.setup(16, GPIO.OUT)
#Output to pin 12 -> turn fan on
GPIO.output(16, GPIO.LOW)
#Wait 20 Minutes
time.sleep(1200)
#Output to pin 12 -> turn fan off
GPIO.output(16, GPIO.HIGH)
#Restart terrariumpi
os.system('systemctl start terrariumpi.service')
```

Is there another way instead of stopping the whole service, pause and restart. 
Is it possilble to controll it via API in /api/environment/humidity -> Setting enabled: false  ?

Ps.: 
Is there a roadmap of what you are planning to integrate in the next version or even a release date? 
Maybe I don't need to develop own scripts if it just takes a few weeks or so...

theyosh, 2020-10-05:
Hm, your script could be more simple like: https://github.com/theyosh/TerrariumPI/blob/master/contrib/toggle_switch.py

That script uses the API, to overrule a power switch (manual mode) and make the power switch go on. Then it waits (now 0.0 seconds) and then toggle the power switch once more, and restore the automatic mode.

This way, you do not have to shutdown the hole system.

And no, I do not have a roadmap. I just make irregular releases and add features which I like / think are usefull. That's it :)

theyosh, 2020-11-01:
Did the above suggestion help? Can we close this issue?

Dave0408, 2020-11-01:
Hello,

yes i have now been able to set it up the way i would like it to be through the external scripts. Thanks for the hint. 
I am very happy so far. 

Nevertheless I would be happy if I could do everything in the software sometime in the future. So I am curious about the new version ;-)

closed 2020-11-01 by Dave0408