theyosh/TerrariumPI #444: [FEATURE]: Logging table of actions

Dave0408, 2020-07-30:
Hello, 

I think it would be quite nice to have the possibility to check in a logging table when which action was executed. 
For example I set a threshold min/max for the humidity. When this is exceeded the fans will go on. If the low threshold is reached the rain system will start. Furthermore, depending on the weather location, the light starts and stays on for a maximum of 10 and a minimum of 8 hours. 

I would like to have a history of the actions performed by the TerrariumPI. 
Something like this: 
[image]

This would be nice to get an overview what happend during the i wasn't sitting next to my terrarium. Yes, there are such times. ;-)

If there is already such a feature let me know.

Dave0408, 2020-07-30:
Ok. Maybe ist just the Switch-> Status page. 
But there is not the reason for action....

theyosh, 2020-07-30:
Hmm, maybe that is already possible. There are logfiles, which log a lot with timestamps. If you go to the System -> Log page and then on the right side the wrench icon. There you can download the log. I just tried to import it in excel with fixed width fields.

I think you can tell excel that the first field is a date. And then you van filer on INFO or WARNINGS. But also all the powerswitch actions.

theyosh, 2020-08-08:
For now, I have to say that this will not be included in the current version. I am at the point that I started a rewrite, and therefore I am only fixing small issues or features.

There is logging at the moment, you could even consider enabling debug logging, and then parse it with Excell of Python to get your data. As you have specific wishes, they are not for everybody. So that means that I have to make a new interface where you can 'query' the logs and get specific output. That is a lot of work.

For now, I leave it up to you to make some small script that format your data or use Excell.

Dave0408, 2020-08-08:
Okay. I can totally accept that. 
I'll see what I can do for my own. Python ist not new for me. ;)

closed 2020-08-08 by Dave0408

theyosh, 2020-08-10:
Python loves CSV and Pandas ;)