theyosh/TerrariumPI #445: Read Temperature from SONOFF device

ewiandr73, 2020-08-01:
Hi,

Is it possible to read the temperature from a SONOFF device with a DS18B20 sensor attached? 

The following url allow me to display the temperature.

**http://192.168.x.x/cm?cmnd=sonoff/status%2010**

This displays the following JSON information in the web browser

StatusSNS |  
Time | "2020-08-02T09:47:50"
DS18B20 |  
Id | "01143FAA03AA"
Temperature | 15.3
TempUnit | "C"

This would save me having to run sensor cables from the RPi.

Thanks 

Andy

ewiandr73, 2020-08-02:
the URL changes to once it loads

http://192.168.x.x/cm?cmnd=status 10

theyosh, 2020-08-08:
Hi,

I think this is possible but untested :).

So, what you could try is to add a new temperature sensor with the hardware type 'Remote' and sensor type temperature. Then you can use the following address (and this is not tested):

`http://192.168.x.x/cm?cmnd=sonoff/status%2010#Temperature`

This should load the temperature status, and return the value temperature back. But this is based on your JSON output here. If this is not working, please provide a screenshot of the JSON formatted in the browser.

More info: https://github.com/theyosh/TerrariumPI/wiki/Remote-data

ewiandr73, 2020-08-08:
[image]

Here is the screen shot, I tried what you suggested but it didn't seem to pick anything up.

Thanks

Andy

theyosh, 2020-08-08:
Ah, that is way I asked for a screenshot. The address is in this case:

`http://192.168.x.x/cm?cmnd=sonoff/status%2010#StatusSNS/DS18B20/Temperature`

You need to walk the JSON path as described in the remote hardware info

ewiandr73, 2020-08-08:
[image]

Works, Thankyou 👍 

Andy

theyosh, 2020-08-08:
Nice.... And more and more devices do support JSON output... and that makes it all that more easy...

ewiandr73, 2020-08-08:
The sonoff can read upto 8 temp sensors, I'll have a play and report back if I can get it to read more than just this one.

theyosh, 2020-08-08:
Hm, that could be a bit problematic. The software is capable of caching multiple sensor read outs per device. But this is only available for devices of which I know. Like BME280, miFlora, DHTXX etc. They have all more then one sensor. So I physically readout the hole device, and store all the different values in memory. This will reduce the load on a sensor.

But for remote sensors, I have not done this. As there is so much different outputs. So that means I will readout the sonoff multiple times per update. It could be that this is then getting to much for the Sonoff or it takes to much time for my software.

I think some caching could be made, but that means that I have to change a lot in the remove sensor code. And at the moment I am at a rewrite of the code, and only small changes will be done. So no caching of remote sensors at the moment.

theyosh, 2020-08-08:
But, I am interested in picture. Can you make one to show how you have connected the temperature sensor. Or do you have a webpage which give that information.

ewiandr73, 2020-08-09:
Hi,

Multiple DS18B20 sensors also work..

This should work for any SONOFF device that is flashed with Tasmota.  
I tested it on the Dual R2 and the 4CH R3, both work.

[image]

ewiandr73, 2020-08-09:
One thing the graph doesn't seem to be logging anything is this normal?

theyosh, 2020-08-09:
Don't think so.

But did you add all the 8 sensors? That could case a timing issue. Could you make a screenshot of the screen session: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

And it could be that it just need a restart...

ewiandr73, 2020-08-09:
I only setup 2 sensors as that's all I have free at the moment.

The problem with the graph seem to be that although I could see the sensor in the sensor status screen there is no data for it in the setting.cfg file.

[image]

ewiandr73, 2020-08-09:
I tried adding one with the screen r running and noticed the following just off the top of the screen.

[image]

theyosh, 2020-08-10:
These errors are strange.... What is your setup?

- Which python version?
- Using the latest version of my software

Because the bluetooth error should only be there with a Pi zero. Pi3 and Pi4 do have bluetooth, and should work.

Maybe rerun the installer once more? The best is to use Python3

ewiandr73, 2020-08-10:
Raspberry Pi 4 Model B Rev 1.1 - TerrariumPI 3.9.8 

I have just re-installed it.
TerrariumPI v. 3.9.8 (Python2)

Funny I though it was using Python3.

Can I just run the following and change /etc/rc.local to update to Python3?
cd /home/pi/TerrariumPI
sudo ./install.sh 3
sudo ./start.sh 3

theyosh, 2020-08-10:
Yup: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35

ewiandr73, 2020-08-10:
seems to be the same.

I tried it on another setup i have. (Rpi4 Python3 Tpi 3.9.8) it also doesn't save anything in the setting.cfg,  and it also displays the temperature in the Gui.

[image]

[image]

[image]

The added sensor doesn't survive a reboot.

theyosh, 2020-08-10:
Crap, this is due to '%' in your address. That is being escaped in the config....

So, try add a second '%' to the address string. Or replace '%20' with a normal space.

ewiandr73, 2020-08-10:
The Space worked, the extra % didn't.

Working Address
http://192.168.86.90/cm?cmnd=status 10#StatusSNS/DS18B20-1/Temperature

theyosh, 2020-08-10:
Ah, ok, I just updated the config to support % characters... but nice... it is working now both ways

theyosh, 2020-08-27:
Still working? Can we close this issue?

ewiandr73, 2020-08-27:
Hi, 
Yeah sorry I was going to test the % again, but been so busy.
Thanks

closed 2020-08-27 by ewiandr73