theyosh/TerrariumPI #446: Setup Question, pgio switch on 2 enviroment alarms (and timer+sensor on switch)

bggangel, 2020-08-02:
Hi,

I have a question if it is possible to have a GPIO switch to be controlled through the environment settings by 2 high alarms.
I have a grow box with a humidifier and a light in it and a bunch of fans on it.
I want the temp to be around 75 and the humidity around 70.
If any of these are to high, I'd like the intake and exhaust fans to kick on.

I set this up by adding the fans switch to the high alarm of the temperature and the high alarm of the humidify.
This seems to work last night, but this morning, the fans stay on anymore. They turned on and then off right away.
I looked at the logs and saw this:
2020-08-02 09:26:01,612 - INFO    - terrariumEnvironment - Environment temperature is turning on the alarm max powerswitches based on sensor
2020-08-02 09:26:01,613 - INFO    - terrariumSwitch      - Changed power switch 'Fans' of type 'gpio' at address '10' from state 'False' to state 'True' (Forced:False)
2020-08-02 09:26:01,680 - INFO    - terrariumEnvironment - Environment humidity is turning off the alarm max powerswitches based on sensor
2020-08-02 09:26:01,681 - INFO    - terrariumSwitch      - Changed power switch 'Fans' of type 'gpio' at address '10' from state 'True' to state 'False' (Forced:False)
So the temp alarm turns it on and the hum then turned it off.
Not sure why this did not happen last night. Maybe this happens now because the humidity is below the minimum alert, but humidity_alarm_min_powerswitches = empty, so that should not happen.

So how do I setup this so both can control the gpio switch?
I did see something about "Forced=False" but am not sure how to set this to true or even of that would help.
And if it is not possible with a GPIO: I've got a merros MSS425E coming on Monday, would it be possible with that?

And lastly in regards to these switches: Is it possible to put a switch on a timer AND based on a sensor? I tried setting this up by setting a switch up twice, but also here, it seems that the relays got confused after a while thinking the switch was on, when it wasn't.
Again, maybe this is a GPIO limitation and would the MSS425E be better for this?
Or am I doing something completely wrong?

Thank again for your work and I am impressed by the fast response time on issues.

Here are the relevant pats of the settings for the turn on/turn off problem(I think) 
[sensor1596231221]
address = 77
alarm = True
alarm_max = 80.0
alarm_min = 60.0
error = False
exclude_avg = False
hardwaretype = bme280
id = 1596231221
limit_max = 100.0
limit_min = 40.0
max_diff = 20.0
name = BME280 Humidity in the back
type = humidity

[sensor1596231361]
address = 77
alarm = True
alarm_max = 80.0
alarm_min = 65.0
error = False
exclude_avg = False
hardwaretype = bme280
id = 1596231361
limit_max = 90.0
limit_min = 50.0
max_diff = 25.0
name = BME280 Temperature in the back
type = temperature

[sensordbc736f70eecf78cbced1ff618526219]
address = c4:7c:8d:6b:54:ad
alarm = True
alarm_max = 80.0
alarm_min = 65.0
error = False
exclude_avg = False
hardwaretype = miflora
id = dbc736f70eecf78cbced1ff618526219
limit_max = 90.0
limit_min = 50.0
max_diff = 10.0
name = MiFlora Left Temp
type = temperature

[sensor1596238116]
address = 22
alarm = False
alarm_max = 80.0
alarm_min = 65.0
error = False
exclude_avg = False
hardwaretype = dht22
id = 1596238116
limit_max = 90.0
limit_min = 50.0
max_diff = 10.0
name = dht22 Temp on the right
type = temperature

[environment]
...
humidity_alarm_max_door_state = ignore
humidity_alarm_max_duration_on = 60.0
humidity_alarm_max_light_state = ignore
humidity_alarm_max_powerswitches = 1595796058
humidity_alarm_max_settle = 5.0
humidity_alarm_max_timer_off = 59.0
humidity_alarm_max_timer_on = 1.0
humidity_alarm_max_timer_start = 00:00
humidity_alarm_max_timer_stop = 00:00
humidity_alarm_min_door_state = ignore
humidity_alarm_min_duration_on = 115.0
humidity_alarm_min_light_state = ignore
humidity_alarm_min_powerswitches =
humidity_alarm_min_settle = 120.0
humidity_alarm_min_timer_off = 59.0
humidity_alarm_min_timer_on = 1.0
humidity_alarm_min_timer_start = 00:00
humidity_alarm_min_timer_stop = 00:00
humidity_day_night_difference = 0.0
humidity_day_night_source = lights
humidity_mode = sensor
humidity_sensors = 1596231221
temperature_alarm_max_door_state = ignore
temperature_alarm_max_duration_on = 120.0
temperature_alarm_max_light_state = ignore
temperature_alarm_max_powerswitches = 1595796058
temperature_alarm_max_settle = 125.0
temperature_alarm_max_timer_off = 59.0
temperature_alarm_max_timer_on = 1.0
temperature_alarm_max_timer_start = 00:00
temperature_alarm_max_timer_stop = 00:00
temperature_alarm_min_door_state = open
temperature_alarm_min_duration_on = 0.0
temperature_alarm_min_light_state = on
temperature_alarm_min_powerswitches =
temperature_alarm_min_settle = 120.0
temperature_alarm_min_timer_off = 59.0
temperature_alarm_min_timer_on = 1.0
temperature_alarm_min_timer_start = 00:00
temperature_alarm_min_timer_stop = 00:00
temperature_day_night_difference = 0.0
temperature_day_night_source = lights
temperature_mode = sensor
temperature_sensors = dbc736f70eecf78cbced1ff618526219,1596231361,1596238116
...

theyosh, 2020-08-09:
Hi,

short answer: you need a individual/unique power switch for every environment part. So it will not work right now.

long answer:
The environment states are also actively checking the state of the power switches. And this means that when the sensor value is below the alarm low, it will not allow the alarm high to be on. And that is now 'your problem'. If the humidity is within its range, then you can toggle the power switch and it will not change. Because the humidity is not to low. But if the humidity is below the low alarm, it will always force the power off.
And the same goes with the temperature part. When it is to cold, it will not allow power on the alarm high switch, which is the same for humidity.

So I think, it is only possible if you have multiple fans with each their own power switch.

Also adding a power switch twice with difference names will not work, as I check on the address values if the power switch is unique. And will not allow duplicate switches based on the hardware addresses.

bggangel, 2020-08-09:
ok, thank you for the clear exploration. You answered all my questions, thanks again for all the hard work and time you put into this project.

closed 2020-08-09 by bggangel