theyosh/TerrariumPI #447: [FEATURE]: Enclosure temperature control triggered by CPU temp

blatogera, 2020-08-08:
Hello, 
I have the lower enclosure for electronics, lamps, water tank etc... and looking to set temperature in there.
Things gets heated up in pretty quick if lamps going on, causing the Pi to overheat.
I installed a fan and right now its controlled by the Terrarium top BME280 temp sensor HIGH alarm ( as a temorary solution)

**Describe the solution you'd like**
Would be great to trigger a switch/GPIO with fan controlled by CPU temerature.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

bggangel, 2020-08-08:
I realize that you are requesting this to be part of the TerrariumPi, but this link describes exactly what you want in a pretty clear way.
[https://howchoo.com/g/ote2mjkzzta/control-raspberry-pi-fan-temperature-python]
Maybe it can help you out in the mean time.

blatogera, 2020-08-08:
@bggangel Thank you, yes i tried that one first, but after installing and rebooting, the Pi could not boot up. Tried several times, i had to recover from a backup image to have it work again.
Not sure why or how it interfere with terrariumPi, as i have zero programming experience :) - Thats why i opened the request

theyosh, 2020-08-09:
Hi,

this is possible in multiple ways. I the easiest is to add a new remote temperature sensor. Use as address:

`http://localhost:8090/api/system#temperature`

This will load the CPU temperature from the system status page. As I already measure the CPU temperature. Then you can use this in the heater environment and in the high value part.

Then add a GPIO (inverse) switch for the fan.

If you are already using the heater environment, then I think it is not possible directly.

Then you could also 'abuse' a moisture environment. As the logic is the same.

theyosh, 2020-08-09:
But maybe using some cooling on the Pi: https://magpi.raspberrypi.org/articles/group-test-best-raspberry-pi-4-thermal-cases-tested-and-ranked

blatogera, 2020-08-09:
> But maybe using some cooling on the Pi: https://magpi.raspberrypi.org/articles/group-test-best-raspberry-pi-4-thermal-cases-tested-and-ranked

thanks, I'm currently using this passive cooling house, which is working good, but as the air is not circulating in the enclosure it cant discipate,

I'll try the mositure section tu use the cpu as remote sensor, 

[image]

blatogera, 2020-08-09:
> http://localhost:8090/api/system#temperature

I can't use the moisute section as I cant select temperature sernsors from the dropdown menu, 
[image]

[image]

blatogera, 2020-08-09:
Never mind, if i set the type to moisture sensor, i can select :) 
Thank you'
works great!

theyosh, 2020-08-10:
Oh crap, that is true... Only one heater is available. In my new version there will be an option to have multiple heating environments. And then it would be easier to do.

But for now, this is also a solution.

theyosh, 2020-08-27:
Ok, can we close this issue? This feature will be possible in my new version that is currently being developed....

blatogera, 2020-08-27:
yes, thank you!

closed 2020-08-27 by theyosh