theyosh/TerrariumPI #448: [Feature/question] PC PWM Fan Control

se7en9057, 2020-08-08:
**Setup:**
 - Python version: [ 3]
 - Raspberry PI: [3B]+

**Describe the bug**
How to control PC Pwm fan using GPIO and Terrarrium pi

**Expected behavior**
able to speed up or slow down fan using PWM wire

**Additional context**
Raspberry pi has hardware and software PWM capabilities on many GPIO pins can this translate into a feature or functionality?

theyosh, 2020-08-09:
This can be done by using a PWM dimmer in the power switch section. I am using something like https://potentiallabs.com/cart/buy-irf520-driver-module-mos-fet-tube-drive-module-arduino-driver-module-online-india (irf50 mosfet)

It can use up to 24 volts at/or 2 amps. So I am using just a 12 volt fan, and this dimmer and it should work.

The only downside is that I am currently relaying on hardware PWM pins. (I think). I am using pin: https://pinout.xyz/pinout/pin32_gpio12#

se7en9057, 2020-08-09:
i see so there is no way to drive the PWM fan directly from the PI's GPIO or Power headders using the max 5v pins. no worries i'm still finding things i'm going to need on this terrarium. lots of little things to automate :) [https://imgur.com/a/nBD5jOe](url)

closed 2020-08-09 by se7en9057

theyosh, 2020-08-10:
Hmm, I am not sure. So maybe give it a try with the DC pwm dimmer. It could be that there is still support on any GPIO for PWM. I am not sure, but I use the GPIO daemon for it, which should be able to make any gpio pin PWM.

But in my new version, I have to drop it, due to the fact that it is getting complicated to explain all the GPIO actions and what is possible etc... So I try to keep it simple :)