theyosh/TerrariumPI #449: received email: "ATTENTION: Service Termination of Your meross Smart Device due to Security Concerns"

bggangel, 2020-08-09:
Hi,

so this morning my Merros MS4225E stopped working, as it was blinking. It had been working fine for a week.
After unplugging a few times, nothing helped. I ended up long pressing the power strip, remomving it from the phone app, and at one point it started working again.
It has been working fine again for the last 8 hours, but I just was checking my emails and saw the following mail from info@notice.meross.com, that was mailed to my 4 minutes before the strip stopped working (I could tell by the log):
**********
Dear Customer,

This is meross cloud security team contacting you.
First of all, thanks for purchasing meross and we hope that you enjoy life with our smart products.
Recently our cloud security system has noticed that your devices are communicating with the could server at an extremely high frequency. We consider these behaviors abnormal and are concerned about the security of your devices.
Due to the security concerns and the intent to avoid any further potential damage, we will terminate the cloud services of these devices for now.
If you are sure that your devices are safe, NOT hacked, and are performing as expected. Please first lower your periodic request rate to no more than one message every ten seconds and then send an email to <cloudservice@meross.com>. We will resume cloud service for you within 24 hours. Please note that any abnormal high cloud request frequency will cause cloud service termination again.
Should you have any information you would like to provide us with or would like to offer us any clarification, please contact us within 24 hours.

Thanks!
meross Cloud Security Team
***************

Well, that sucks.... of course I sent an email back that they should not do this again (the strip controls my humidity).
I did not get a response and do not have high hopes I will get one...
But in case they decide to do this again, I'd like to know: When is TerriumPI accessing the cloud? I assumed it was just on startup to scan for the switches. Or does it contact every time a switch status is changed?
I don't know yet what "to much" is for Merros, but knowing when the cloud gets accessed is a start.
Maybe my humidity controls are set to tight, but from what I can see, it switches about every 3 minutes. That does not seem like "to much"....
Anyway, was curious what you were thinking about this and at least wanted to inform you that this apparently can be a thing.

bggangel, 2020-08-09:
ok, after thinking about it for a while and checking, I noticed that I had left a setting from an experiment, which probably kept switching on and off the fans attached to the strip every few seconds. 
theyosh has already explained to me in another issue what I did wrong there and in the mean time I had updated that setting already, without knowing it did this during the whole night. 
So I think it won't happen again, but it is good to know that "they are watching"... :)

bggangel, 2020-08-09:
leaving this open for a day, so it can get some eyeballs and people can learn from my mistake. Will close tommorow.
@theyosh : no action from your side needed.

theyosh, 2020-08-10:
Hi @bggangel 

I think I can explain this. My software will make a connection every 30 seconds to check the current states. This is done so that you can also toggle a power switch with your app, and my software will 'see' that and update its state according to it.
So that could trigger an alarm, every 30 sec a new connection.

I know this is not the right way to do it, as this switch uses MQTT and therefore, I should just subscribe to a channel once, and just listen for messages. But as I do not have this hardware device, I cannot test various ways of coding. But then again, I use https://pypi.org/project/meross-iot/ which does not support MQTT.

So, if it is an error that my software makes to much connections, I can drop the support for checking the state every 30 seconds. That means, that my software will NOT see any external actions, and it will not check the actual state of the switch anymore.
Then there are only hits to the mereoss cloud when the change the power switch.

> Please first lower your periodic request rate to no more than one message every ten seconds

I am not sure if we hit this. I think not. As far as I know, I will only hit the cloud once every 30 seconds, and when there are actions. As there are more users using this power switch and you are the first with this issue, I think that my software is withing the limits when you toggle once every minute (max). So I would expect that this is happening due to a setup that is switching to much.
But I have to say, one messages very 10 seconds is pretty fair and should be enough.

theyosh, 2020-08-10:
Hmm, it is a 'known issue': https://github.com/albertogeniola/MerossIot/issues/112

bggangel, 2020-08-10:
"But I have to say, one messages very 10 seconds is pretty fair and should be enough."
I agree, I just wish they would warn then terminate later, not the other way around.
It probably was because of a wrong "on/off" setup... I hope. 

Before reading your response, I was thinking to create some more relay/gpio plugs. But if I am the first one with this message on your group, I will give the strip one more shot.

closed 2020-08-10 by bggangel