theyosh/TerrariumPI #450: IRF520 not outputting voltage

se7en9057, 2020-08-12:
Im sorry to even bother you with this however, I have tried to set this up best i know how. sig / gnd from pi on GPIO 32 and GND

GND / VCC are from 12v pwr supply and i can confirm i'm getting 12v. 
V+/- on the output side is getting milivolts at best. fan isn't spinning. 

I'd like to use this for fan speed control as well as LED control. 
[https://imgur.com/a/TG0yi5G](url)

theyosh, 2020-08-12:
You need to connect the vcc to a 3.3 volts pin. I think a small led should light up when the board is powered.

It needs ground, 3,3 volts and a PWM pin to work. So three connections. That should do the trick

se7en9057, 2020-08-12:
hmmm i think ive tried that but i'll give it another go.

se7en9057, 2020-08-12:
ok so i'm on pin 32 which should be PWM0 on the GPIO. confirmed 3.3v on the vcc pin but i was looking at the schematic and it almost looks like that pin goes nowhere but i could be wrong. it got 12v coming in . but still no love on the output side. 
[image]
https://pinout.xyz/pinout/pin32_gpio12

theyosh, 2020-08-27:
Hmm, then I have no Idea.

I used three wires from the Pi to the dimmer.

1. Ground
2. 3.3 Volt
3. GPIO on pin 32

So your screenshot looks ok about the address. Also the wires of the 12 volt look ok on the first images.
Then when I change the dimmer value in my software, I see a light lit up at the dimmer. So every value above 0 will give a led on the dimmer, that it is working. So if you do not see that led light, it could be that there is something wrong with the dimmer or Pi.

Then there is only one way to check. Use a second Pi, and make your own python code to see if it works. I do not have an example script that you can use for testing. The idea is that when there are hardware problems, you test the hardware separately without my software, to make sure the hardware itself works.

se7en9057, 2020-08-30:
@theyosh roger that i'll take a look and if i find an issue i'll re-open / open new issue. thanks for responding :) enjoying your software

closed 2020-08-30 by se7en9057