theyosh/TerrariumPI #451: [BUG]: Error after updating

PhoenixNP, 2020-08-27:
**Setup:**
 - Python version: 3
 - Raspberry PI: 3B

**Describe the bug**
After starting the service it gives this error(s):

Aug 27 09:20:21 terrarium01 python3[832]: Traceback (most recent call last):
Aug 27 09:20:21 terrarium01 python3[832]:   File "/home/user/TerrariumPI/terrariumEngine.py", line 643, in __engine_loop
Aug 27 09:20:21 terrarium01 python3[832]:     self.power_switches[power_switch_id].update()
Aug 27 09:20:21 terrarium01 python3[832]:   File "/home/user/TerrariumPI/terrariumSwitch.py", line 231, in update
Aug 27 09:20:21 terrarium01 python3[832]:     data = self.get_hardware_state()
Aug 27 09:20:21 terrarium01 python3[832]:   File "/home/user/TerrariumPI/terrariumSwitch.py", line 476, in get_hardware_state
Aug 27 09:20:21 terrarium01 python3[832]:     data = subprocess.check_output(cmd).strip().decode('utf-8').split('\n')
Aug 27 09:20:21 terrarium01 python3[832]:   File "/usr/local/lib/python3.5/dist-packages/gevent/subprocess.py", line 404, in check_output
Aug 27 09:20:21 terrarium01 python3[832]:     raise CalledProcessError(retcode, process.args, output=output)
Aug 27 09:20:21 terrarium01 python3[832]: subprocess.CalledProcessError: Command '['/usr/bin/sispmctl', '-d', '-1', '-n', '-g', '4']' returned non-zero exit status 2

theyosh, 2020-08-27:
Hi, not sure how long it was between the updates, but I think this will be fixed by running the installer again

`sudo ./install.sh 3`

I think you are missing a debian package `sispmctl`

PhoenixNP, 2020-08-27:
I fixed it in the meanwhile =)
sispmctl package has been on the system for a long while, however because of the time in between the updates I had to re-adjust to the changes you made on the EG-PGM.
Wrote a script to get the correct serial-number from the EG-PGM and replace the "old" config lines with the new expected config lines =)

PhoenixNP, 2020-08-27:
Quick fix for most people would be just to strip the EG-PGM from the config, reboot the Pi and re-order your switches (and put them in environment again)

closed 2020-08-27 by PhoenixNP

theyosh, 2020-08-27:
Yeah, I was afraid that this could be a side effect of my code. Sorry :)