theyosh/TerrariumPI #452: Interface Can't Be reached

Nachomachoman, 2020-08-29:
Hello. I followed all the steps to install the TerrariumPI and everything and now for whatever reason, I can not get into the interface module. I recently did a fresh install of Rasbian and everything was working perfectly fine then. Now I get this "Site cannot be reached" *ip address* refused to connect? I followed a few help topics here but nothing seemed to help. I even tried to completely disable my network firewall and still the same issue is happening.

theyosh, 2020-08-30:
Hi,
So I can assume you are using Python3? 

Could you see if there is a screen session running: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

If not, what happens when you manually start it:

`python3 terrariumPI.py`

Also are you using the port number :8090 to the ip address in the browser address bar?

Nachomachoman, 2020-08-30:
As far as the screen session it says there is no screen session running. When I do python3 terrariumPI.py it says there is a server running at *myip address*:8090 when manually starting. And yes I am using :8090. 
[image]

And yes I am using python 3

theyosh, 2020-08-31:
Hmm, if you are using Python3, then you should start it as:

`sudo ./start.sh 3` as the 3 specifies the python version. And this message is not enough. 

So, I would like to ask to run the following command and send another picture with the output:

`python3 terrariumPI.py` in the same folder as you run the start command.

Matze75-Swiss, 2020-08-31:
Hi Guys,

I have exactly the same problem. Tried with Python2 and Python3 as well, always the same result.
When running screen -r it seems that the Adafruit_DHT module is missing...

[image]

Thanks for helping...

theyosh, 2020-08-31:
Hmm, strange... what if you run the following command:

`sudo pip3 install Adafruit_DHT`

Does this install the library? The results should look like:

```Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting Adafruit_DHT
  Downloading https://www.piwheels.org/simple/adafruit-dht/Adafruit_DHT-1.4.0-cp37-cp37m-linux_armv7l.whl
Installing collected packages: Adafruit-DHT
Successfully installed Adafruit-DHT-1.4.0 
```

Matze75-Swiss, 2020-08-31:
Ah, great, now it's running. There was another missing module BME280 - installed it also, now it runs.

Great stuff!!

Another question: Does the module support EZO pH and Conductivity sensor - I want to use it for a Dwarf Caiman?

theyosh, 2020-08-31:
You mean this one: https://atlas-scientific.com/circuits/ezo-ph-circuit/ It has I2C, so yes, that should be possible.

But I do not own this kind of hardware, so that means I need some help with testing. As you figured out another missing library, I think you have some experience with git? Because then we could test this in a separate branch

Nachomachoman, 2020-08-31:
[image]

This is what I got after running python3 terrariumPI.py

theyosh, 2020-09-01:
Hmm, you are also missing the BME280 module. This can be installed manually:

`sudo pip3 install bme280`

Then try again `python3 terrariumPI.py`

If then is running, you can press `Ctrl+C` and then start it normally:

`sudo ./start.sh 3`

And it should all work. If there are still issues, you can also try to rerun the installer

`sudo ./instal.sh 3`

This will rerun the installer, and if there are more missing modules, they will get installed

Matze75-Swiss, 2020-09-01:
Yes I mean the Atlas Scientific. Actually there is a quite good Library for the Raspberry Pi Tentacle shield here: https://www.whiteboxes.ch/docs/tentacle/t3/#/code I just installed it and it seems to work, but no idea how to integrate in TerrariumPi HMI. I come from Arduino and have not so much experiences with python and raspberry pi yet. 
The Atlas EZO is normally pretty simple: you can send a request command "r" to the sensor address and the sensor returns the measured value. Furthermore there is a calibration command to calibrate the sensors from time to time; for calibration the sensor value is read until it is stable and then the calibration command is send which normally consists of the command "cal" and one or two further parameters separated by comma; combinations could be (cal,clear or cal,low,value or cal,mid,value or cal,high,value, ...). within reef-pi they have a simple interface where you can put the values in and have an execute button for each calibration step. When referring to the Tentacle shield we need three values; typically pH, Conductivity (or Dissolved Oxygen) and Temperature.
But to answer you question: yes, I would be very happy developing this together with you. I could provide some sketches how the interface should be and test this.  

May be we should move this in a separate topic?!

Nachomachoman, 2020-09-01:
Yay! Works perfectly now. Thanks!!

theyosh, 2020-09-02:
@Matze75-Swiss yeah, make a new issue per sensor you would like to use. 

The calibration part, is a bit hard to make in my software. So that would be something you have to do outside my software. But I can add fields to the sensor, so you can set those values in my software.

Also, when opening a new issue, could you also provide a screenshot where it shows the output of a read command? As I can't find it, I do not know how the data is returned. I do not see a separate function for PH or Temperature. So I guess all data is at once.

theyosh, 2020-09-02:
@Nachomachoman nice that it is working now.

tassieKev, 2020-09-05:
Just reporting the same issue as the OP. Fresh install on Pi3 with Rasp Lite image. BME280 module not installed by installer

[image]

but installed with pip3 

`pi@raspberrypi:~/TerrariumPI $ sudo pip3 install bme280
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting bme280
  Downloading https://files.pythonhosted.org/packages/4f/01/8005507cce43f77170981b81c92c8114433f88e74bba4123f639f2ff264e/bme280-0.5-py2.py3-none-any.whl
Installing collected packages: bme280
Successfully installed bme280-0.5`

And running fine after that :)

theyosh, 2020-09-05:
Ok, this is an issue with pip services, which is outside my reach.

So, then the only solution is run the installer once more (re-run) or manually install the missing modules.

I cannot fix installation issues, as I am relaying on external services.

Donpilkerton, 2020-09-10:
Hello, Any ideas on why my interface would not be available after restarting my pc. NO changes were made just a shut down over night. I am able to ssh into the pi and I am able to log in and get the MOTD but I am not able to go into the web browser and access the GUI. Was working fine yesterday. This is over both LAN and WLAN interfaces

theyosh, 2020-09-19:
Closing now. I will keep in mind that I have to look into the BME280 installation..

closed 2020-09-19 by theyosh