theyosh/TerrariumPI #453: terrariumSwitch turns on light due to old setup

Dave0408, 2020-09-05:
Hello, 

i've got a little problem with my light settings. When I started using TerrariumPi i configured an interval between 08:00-18:00 to switch on my led light using the timer function of the switch. I changed to using an environment depended light setup. 
What i observe now is TerrariumPi turn on the light at 08:00 and turn it off immediately. My environment setup starts at ca 08:46 for 10 hours. So it stays on longer than 18:00 and i can't see any false behaviour at 18:00. 

How can I avoid that short on/off situation at 08:00? It seems he somewhere rembered my old config. 

Thanks in advance.

For further analysis:
Log:

> 
> 2020-09-05 08:00:21,733 - INFO    - terrariumEnvironment - Update done in 0.30227 seconds. Waiting for 14.69773 seconds for next update
> **2020-09-05 08:00:21,549 - INFO    - terrariumSwitch      - Changed power switch 'Skylight LED' of type 'mss425e' at address '1' from state 'True' to state 'False' (Forced:False)**
> 2020-09-05 08:00:21,450 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
> **2020-09-05 08:00:21,443 - INFO    - terrariumEnvironment - Environment light is turning off the alarm min powerswitches based on weather**
> 2020-09-05 08:00:08,696 - INFO    - terrariumEngine      - Update done in 6.51364 seconds. Waiting for 23.48636 seconds for next update
> 2020-09-05 08:00:08,431 - INFO    - terrariumSwitch      - Updated gpio-inverse power switch 'Lüfter' status in 0.00017 seconds
> 2020-09-05 08:00:08,326 - INFO    - terrariumSwitch      - Updated mss425e power switch 'USB Stecker' status in 0.17813 seconds
> 2020-09-05 08:00:08,149 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
> 2020-09-05 08:00:08,043 - INFO    - terrariumSwitch      - Updated mss425e power switch 'Pumpe' status in 0.18236 seconds
> 2020-09-05 08:00:07,863 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
> 2020-09-05 08:00:07,756 - INFO    - terrariumSwitch      - Updated mss425e power switch 'Beregnungsanlage' status in 0.18351 seconds
> 2020-09-05 08:00:07,573 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
> 2020-09-05 08:00:07,467 - INFO    - terrariumSwitch      - Updated mss425e power switch 'Skylight LED' status in 0.45037 seconds
> 2020-09-05 08:00:07,463 - INFO    - terrariumSwitch      - Power switch changed from False state to new True state
> **2020-09-05 08:00:07,302 - INFO    - terrariumSwitch      - Changed power switch 'Skylight LED' of type 'mss425e' at address '1' from state 'False' to state 'True' (Forced:False)**
> 2020-09-05 08:00:07,207 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED
> 2020-09-05 08:00:07,017 - INFO    - protocol             - Current client connection status is: ClientStatus.SUBSCRIBED

Switch config in Web GUI:
[image]

Switch config in settings.cfg file:
[image]

Environment setup:
[image]

theyosh, 2020-09-05:
If you have enabled an timer on a switch, you cannot use that switch in the environment. That will give strange behaviour.

So, I think, you should disable the timer at the power switch, and it should work.

Dave0408, 2020-09-05:
As you hopefully can see there is no current timer configured. It was in the past.

theyosh, 2020-09-05:
Hm, could it be as simple as a restart? It could be that I do not unload the timers of the switches. https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-start-terrariumpi

If it is working after a restart, then this is a bug.

Dave0408, 2020-09-07:
Unfortunately it is not that simple. I tried several restarts of the whole pi.

theyosh, 2020-09-07:
Hmm, could it be that there is also an timer in the Meross app? Because the first on toggle happens after around the line:

 2020-09-05 08:00:07,467 - INFO - terrariumSwitch - Updated mss425e power switch 'Skylight LED' status in 0.45037 seconds

So, I get the felling that there is other software that makes the switch toggle. My software will then update to the new state (on). And then the environment part puts it back out, due to the setup.

I know meross does support timers in the app.

Else, could you provide a logfile from a day?

Dave0408, 2020-09-08:
That was the right hint. The meross app supports timer aswell. I resetet my switch and now everything works fine. Thank you very much.

closed 2020-09-08 by Dave0408