theyosh/TerrariumPI #454: DHT22 AND DHT11 sensors reporting null values.

Donpilkerton, 2020-09-09:
I have updated any and all of the DHT libraries that I can find on get, and I am still unable to get the sensors to read properly. When hooking them up to a Arduino Uno I am able to see the temp and humidity read outs but I am unable to see anything after hooking them up to the Pi that is running Terrarium. This is kind of unrelated to the initial bug report but I am unable to save any settings under the system tab in the environment or settings sections. Please let me know if you need any more information.

closed 2020-09-09 by Donpilkerton

theyosh, 2020-09-11:
Hmm, what was the issue? As I had no time to look into it.

But look at https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed+dht

And I suggest to use BME280 in stead of DHTXX for the future.

Donpilkerton, 2020-09-11:
Yep, after taking a look around the closed issues I saw that it was a pretty common issue. I ordered a  BME280 and it works like a charm! I closed the issue sorry for not updating what I found to work!