theyosh/TerrariumPI #455: Pi Zero Install

cyber-shifter, 2020-09-09:
I realize that Pi Zero isn't officially supported so this is more of a request for everyone else, and less for the developer. I have too many tanks to get a Pi 3 for each one, and the Pi Zero is at least half the cost, if not more. So I've got to try to make it work. I realize that it has to run as root to access the GPIO. I don't have any sensors hooked up yet, I'm just trying to get it to start. The only feedback I'm getting from `screen -r` is that it can't find the bme280 module. Is that how it will present the problem of not running root? Or is this something different?

Dave0408, 2020-09-09:
That's something different.
Try `pip install bme280`  or `pip3 install bme280`.

theyosh, 2020-09-11:
Hmm, again, it is the bme280 module. And the strange thing is, apparently the module bme280 is working. But I use the module RPI.bme280 in the code. So it looks like both modules do work :)

But this is not related to the PiZero. The problem is that you need to run the software as root user, else you will not get all the GPIO data from all the sensors. Why? that is unclear to me still.
And in my opinion, the zero is lacking CPU power to use it as a developer system. It goes to slow to test it for me. That will take a lot of time to do testing. And therefore, I currently advise to use a Zero at own risk :)

cyber-shifter, 2020-09-18:
Yeah, I know. Unfortunately I don't have the money to buy Pi 3's for all my enclosures at the moment. I will probably do it gradually. Anyway...

So, I attempted to change the owner and account to run the install as to root, but that didn't make a difference. I did a `sudo su` first, then I called the script like normal. It still showed the normal user. I did:

`sudo chown root install.sh
sudo chmod +s install.sh`

to the file, and then called it again, and it still did the same thing. Do I need to just install it with the regular user, then change the owner, etc after the install?

theyosh, 2020-09-19:
Euh, yes this is possible, but you have to change the owner of the folder.  https://github.com/theyosh/TerrariumPI/blob/master/start.sh#L23

So, try: `chown root .` in the /home/pi/TerrariumPI folder. This will make the folder owned by root, therefore if you want to make changes in that folder, you have to do that as root now, and not the normal pi user.

cyber-shifter, 2020-09-22:
Ok. So, I did the install with the regular user. Then I attempted to change the owner of the folders, and also who executes the start.sh file. It did not work. I did it as root using `sudo su` and also did it as my user with the sudo prefex. The ownership is still not changing. And of course the system won't start. I know you can give a password to the root user, and log in as them, but I would prefer not to do that, unless it is the only way.

theyosh, 2020-09-25:
If you use the start.sh script, it does not matter if you start it as root, it will drop to the user how owns the TerrariumPI folder.

But, if you want to test if it will run with root rights, you can also try: `sudo python3 terrariumPI.py`

This will just start the software as root user. That should work. But then you cannot close the terminal. For testing it is ok.

kebs19, 2020-09-28:
I run my system on a Pi ZeroW and it works fine. I just installed and tried everything using a RPi 2B, and swapped the cards.
Using a ZeroW has some advantages, it runs cooler (so I didn't have to use a fan inside the controller cabinet) and uses way less energy (which for a system running 24x7 is important). 
My system is running for more than a year and the only issue I have is that from time to time the sensors stop to work, and I need to turn the system off and back on. It can happen in a week, in a month or can stay on for months... It's ok by me.

cyber-shifter, 2020-09-29:
> If you use the start.sh script, it does not matter if you start it as root, it will drop to the user how owns the TerrariumPI folder.
>
> But, if you want to test if it will run with root rights, you [...]

So, I made "root" the owner of the whole folder, before I did the install. I verified that it took the change. Then I ran the install from root by doing `sudo su` first, and then running the command. It still said it was installing as "pi" so I figured I would try it, and  change the ownership later. Everything installed. When I check the ownership of the folder after the install, it all belongs to "pi" again. But it is running, complete with MOTD. When I go to change settings, and enter the user credentials, they don't work. No iterations work. So I'm not sure what went wrong here. Should I try stopping the terrariumPI service, changing the owners of everything once again, and rebooting it?

theyosh, 2020-10-05:
Ok,

again, for testing, just run `sudo python3 terrariumPI.py`. Does this work? If not, you can stop. Because that means that even running it as root, it will not work.
That all, nothing more....

cyber-shifter, 2020-10-07:
My apologies, I misunderstood what you meant the first time you said it. I will test it and let you guys know how it goes.

theyosh, 2020-10-10:
When it does work as root, the next step is that you first start a screen session by hand. This can be done either as root or normal pi user.

So as a normal pi user do the following:

`screen`
`sudo python3 terrariumPI.py`

then enter Ctrl+A, D and it will detach. Now it will keep on running as root user in a screen session, so you can then close the terminal/ssh session

cyber-shifter, 2020-10-14:
Got it. I'll try this next.

cnelsonsic, 2021-01-24:
Just as an additional datapoint:

I tried this with latest raspbian and terrariumpi, on a raspberry pi zero w, and it seemed to work well enough for me. I've not tried it with a plethora of sensors but it started and ran fine for me.

theyosh, 2021-02-01:
There are some GPIO rights issues with a Pi zero. Running it all as root should fix that, but I do not like that, as that is less secure.