theyosh/TerrariumPI #456: [BUG]: Unable to save settings under system settings tab.

Donpilkerton, 2020-09-09:
I was just wondering what might be preventing me from making and changes and saving them to the systems settings tab. I am able to input the data that I need but when hitting submit after I get a "data could not be save" error. Please let me know if you need any more information from me! Thanks!

theyosh, 2020-09-11:
Hmm, unfortunately, with this screenshot it will be a guess. Is it possible to make a screenshot from the form with the information filled. You should take care about the weather url data, that will contain your APP key. So clear that with a draw program.

Then, did you 'logged in' while you loaded the settings page? You should get an authentication box from your browser to enter a username and password. Did you do this?

And final, when you want to save the form, you can also use the screen debug to see what is happening when you press the save button: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

Dcreeron, 2020-09-20:
I'm getting a similar error. Was originally running 3.9.7 (with full Raspbian and Ux) with no real issues. Tried updating to 3.9.8 and had problems getting it to run. Ended up reformatting my SD card with the latest 'lite' version. Now T-PI 3.9.8 runs but I can't edit the settings:

[image]

I am logged in -- was prompted on the settings page.

Doesn't matter what I try to change. It's always bringing up the error that I need to select a value for the sound card. When I try to select one (headphones or HDMI) it just gives an error.

No errors show up in the output when using "screen".

I've tried to overwrite the defaults.cfg and rebooting but that doesn't work for some reason (maybe the file is locked?).

theyosh, 2020-09-23:
Hmm, this is getting strange. So, what give the following url as output:

http://[terrariumpi]:8090/api/audio/hardware

This should give all the available sound hardware. Could load the url and make a screenshot?

Also, in your picture, I can't see the choice you made for the soundcard. As I do not have a HDMI choice. Are you maybe revering to the raspi-config? That is different I guess..

Dcreeron, 2020-09-23:
When I go to the requested URL this is the response I get:

{"audiohardware": {"bcm2835 Headphones": {"hwid": 0, "name": "bcm2835 Headphones"}}}

I don't have the HDMI cable plugged in any longer, so I'm not getting the HDMI choice. The Rasp-Pi is in another room near the terrarium.

When I had the HDMI plugged in, the settings screen would come up with no sound card selected -- it said "Select an option". It didn't matter which option I picked, headphones or HDMI, it always gave the error.

This morning when I tried, as shown above, the only option was the headphones. I picked the headphones and changed some of the other settings (temp, distance, volume, etc.) and hit SAVE. It no longer gave the error about the sound card setting, but it did complain about the missing weather url. So I filled that in, hit save again, and it worked!

I'd say these problems are related to having the HDMI cable plugged in. That's the only difference.

theyosh, 2020-09-25:
Hmm, that is annoying. I have not tested with a HDMI connected, as I work headless.

But could you do a small test for me? Connect the HDMI again, and load the url http://[terrariumpi]:8090/api/audio/hardware again. Post the outcome here, and disconnect your HDMI. So I can see what the results are when a HDMI device is connected.

Then I can probably fix this also with HDMI connected.

theyosh, 2020-09-25:
Also, change your weather url: #417

Dcreeron, 2020-09-26:
I plugged the PI back into an HDMI cable and I get this for the response:

"audiohardware": {"bcm2835 HDMI 1": {"hwid": 0, "name": "bcm2835 HDMI 1"}, "bcm2835 Headphones": {"hwid": 1, "name": "bcm2835 Headphones"}}}

I'll look into the other weather service, but I don't feel like registering right now. The weather field is required on the SETTINGS page, so I can't delete it (i.e., for NO weather).

Our usage of TerrariumPI on my son's Gecko tank is pretty simple. Mainly using it for the web UX and the temp and humidity sensor I wired up. Haven't gotten around to doing anything with the lights or really anything else.

The weather fields being required makes it impossible to just update other fields (like C vs F, etc.).

Thanks!

theyosh, 2020-10-05:
Sorry for the delay. I have removed the weather requirement in the settings page. So empty weather should work. But I have not tested this will.

And with the audio hardware you have posted here, is working for me. So I do not know what is causing it not to work for you.

theyosh, 2020-11-01:
Is this fixed? As I cannot reproduce...

Dcreeron, 2020-11-01:
> Is this fixed? As I cannot reproduce...

Yes, we were able to save with no weather and the audio issue doesn't seem to be an issue any longer.

Thanks.

closed 2020-11-01 by theyosh