theyosh/TerrariumPI #457: Temperature Sensor "null" return

lukaskom1, 2020-09-15:
Hi,
The TerrariumPi environment looks very nice but I spend 3days to configure the sensors. 
Raspi OS was 4 times installed and TerrariumPI was updated many times. 
DHT22, DHT11, AM2302 were used with the same effect no reading the value
I tried all tips mentioned on the TerrariumPI  Issue tab. 
Using Adafriut library works fine but response time is quite long -30sek.

[image]

Please help me with this task

theyosh, 2020-09-16:
Euh, if you see that it takes 30 sec to readout the sensor, then I say the sensor is broken. It should take no more then 2 seconds to get the values.

Also, I suggest to start using BME280 and drop the DHT. There are a lot of issues with the DHT sensors: https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed+DHT

lukaskom1, 2020-09-16:
I have The BME280 and I connected this with the same result, but it is questioned how I should address this sensor e.g SCL = GPIO2 SDA=GPIO3. How the address should be. please help with this 

BTW: Arduino Uno with DHT return value each every secound

[image]

theyosh, 2020-09-19:
For the BME280, it should be visible on the I2C bus: https://github.com/theyosh/TerrariumPI/wiki/Hardware#i2c-bus

And then you have to enter the address what is shown in the matrix. There should be a value of 76. So enter that value as address in my software, and it should work.

Why the DHT does work on an Arduino, I do not know. Different OS, different libraries and different code is to much to compare.

theyosh, 2020-10-05:
Any success with the BME280 sensor?

lukaskom1, 2020-10-05:
Hi. im not testing
Now im switch on arduino platform with LCD and cloud visualisiation.
In my house home autoimation based on PLC and i have plan to communicate
each other.

after ending i wilk put the code on github

theyosh, 2020-10-05:
Is it then ok that I close this issue?

lukaskom1, 2020-10-05:
ok no problem

closed 2020-10-05 by theyosh