theyosh/TerrariumPI #458: [FEATURE]: Support for MEROSS MS100 sensor

Neamh, 2020-09-15:
As MEROSS switch boards are already supported through the cloud login, maybe it is possible to support the MS100 sensor for temperature and humidity as well.

https://www.meross.com/product/46/article/
https://albertogeniola.github.io/MerossIot/quick-start.html#reading-sensors

theyosh, 2020-09-19:
Hmm, currently the implementation of the meross library is outdated. But I am unable to do any testing, as I do not have the hardware. So at this time, I am in doubt if I can add this in the current version.
Because this will also affect the switch bar as that is using the same library.

I am working on a new version, and there I could possible add it. For now, I have to see to think a bit more. Or do you also own a MSS425e (Smart strip)? Then we could maybe test it on a separate branch.

Neamh, 2020-09-19:
Hi.
I absolutely understand.
I currently have a MSS425F. Does this work as well? It is the same switch but with four outlets.

theyosh, 2020-09-20:
Hmm, if you have that one up and running, I think it should be working....  https://github.com/theyosh/TerrariumPI/blob/master/terrariumSwitch.py#L1272

So, if you restart my software, it will not find any power switches? Aren't there any messages during startup?

theyosh, 2020-09-20:
Ok, if you want to test this, there are a few points to take care off..

1. Get a new Pi, so you can test it without interfering with your current working setup.
2. Clone the software: `git clone --recursive https://github.com/theyosh/TerrariumPI.git` . This is different than the normal install!!
3. Then switch to the meross branch: `cd TerrariumPI` and then `git checkout meross`
4. Run the installer: `sudo ./install.sh 3`

Then you should have a new setup, with the latest Meross code. Your powerswitch should work with that. There is no sensor code yet. I first want to see if the power switches still work with the new code before adding new sensor code. As they share the same python library

theyosh, 2020-10-05:
Any time to test this?

Neamh, 2020-10-11:
HI, I am sorry... no time for tests so far. Even more, it is currently more or less impossible to get the MS100 here in Germany.

Neamh, 2020-10-21:
Now I had the chance to test the Meross branch. Sadly, it wouldn't detect anything from Meross. No entry in the logs and the power switches are not detected.

theyosh, 2020-10-21:
Something in the screen session?: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

Neamh, 2020-10-21:
Yes, this is the special branch. However, in the screen session I found that there is a login error to Meross cloud. Sadly, this sends the software in a restart-loop where it exits after 5 retries. Can I somehow enter the correct password somewhere without the web interface?

Neamh, 2020-10-21:
Found that I could simply edit settings.cfg. However, the Meross login is in there twice in [terrariumpi] section and [meross_cloud] section.

Neamh, 2020-10-21:
This what I get after correcting the login and a reboot:

`2020-10-21 11:45:47,974 - WARNING - device_factory       - Device Smart Hub (msh300, uuid 1912189057482226280548e1e913cc4f) reported ability Appliance.Digest.Hub. Assuming this is a full-featured HUB.
I've found the following devices:
- Terra AZ Leiste 2 (mss425f)(2007167279735190820048e1e925873b): OnlineStatus.ONLINE
['__annotations__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_abilities', '_abilities_spec', '_channel_togglex_status', '_channels', '_execute_command', '_fire_push_notification_event', '_fwversion', '_hwversion', '_last_full_update_ts', '_manager', '_name', '_online', '_parse_channels', '_push_coros', '_type', '_uuid', 'async_handle_push_notification', 'async_handle_update', 'async_toggle', 'async_turn_off', 'async_turn_on', 'async_update', 'channels', 'check_full_update_done', 'dismiss', 'firmware_version', 'hardware_version', 'internal_id', 'is_on', 'last_full_update_timestamp', 'lookup_channel', 'name', 'online_status', 'register_push_notification_handler_coroutine', 'type', 'unregister_push_notification_handler_coroutine', 'update_from_http_state', 'uuid']
[<meross_iot.controller.device.ChannelInfo object at 0x631d60d0>, <meross_iot.controller.device.ChannelInfo object at 0x631d6070>, <meross_iot.controller.device.ChannelInfo object at 0x631d6110>, <meross_iot.controller.device.ChannelInfo object at 0x631d6090>, <meross_iot.controller.device.ChannelInfo object at 0x631d6930>, <meross_iot.controller.device.ChannelInfo object at 0x631d6910>]`

theyosh, 2020-10-21:
Hmm, is this on a different pi? As i need more info, and the exact error message.

So, I would like to ask to stop my software https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-stop-terrariumpi and then run it manually:
`sudo python3 terrariumPI.py`

And then copy all the output in the console and paste it here within the 'code' block so it will not change the text.

theyosh, 2021-04-10:
[image]

Yesh!!! Party time... Finally after a year.... it is here!! If you have time, could you have time to test this: #539

commit 5816f655 (2020-09-21): Small fix #458