theyosh/TerrariumPI #459: Environment not letting me use switch without timer

cyber-shifter, 2020-09-22:
The subject kind of says it all. I'm setting up an environment for snakes, which is regulated by a temp sensor on the hot side. I want it to kick on when the temperature is too low. I set the values, but when I go to save it will not permit me to save without a time value in the timer fields. Even though I am not using the timer. This is running on a Raspberry Pi 4. Please see the included screenshot.
[image]

cyber-shifter, 2020-10-07:
I'm still having trouble with this. I'm not sure if there is a workaround, or what. I can set both the start and stop to 12:00 AM as a work around, but it still wants me to put a number of minutes in the "Timer On" and "TImer Off" fields. I didn't even select timer, I told it "Sensor".

theyosh, 2020-10-08:
My apologies... The strange thing is that when you select sensor mode, it should make the timer start and stop fields read only. And that is not the case in the screenshot above. With that, it should just fill in some time settings, just to make the form valid.
So I get the feeling there is a javascript error. Have you tried clearing your browser cache? Or could you load your javascript console of your browser to see if there is an error when loading this page?

[image]

cyber-shifter, 2020-10-13:
I'm sorry, work has been crazy lately. I made time this afternoon after work to look into this and some other things, so I will return with data on this tonight. Thanks for the idea! I wouldn't have thought of it.

cyber-shifter, 2020-10-14:
Looked at this last night. There was some kind of weirdness it seems, with my Javascript. I run Linux natively, so I see oddness sometimes just because of that. But once I cleared the cache, it seemed to work right. It still showed those fields as white, when I first set it up. But when I navigated back to that page from the dashboard, it looked like it was supposed to. Thanks for the help on this! I'll go ahead and close the issue now.

cyber-shifter, 2020-12-15:
This is still happening. I'm not seeing any oddities this time. I cleared the cache even. This is very problematic because it won't even save the environment setting when I DO put values into the timer fields. I can't run the environment this way. I need it to function without the timer.

cyber-shifter, 2020-12-15:
I should note that this condition only happens as soon as I select the switch I want to use. The page looks correct  before I do that, with the proper things greyed out. But once I select the switch, it activates the time fields, and it won't proceed further. It's like it's calling the wrong function or something.

cyber-shifter, 2020-12-15:
Oh! I forgot to provide browser details. I tried this in both Chrome and Firefox, version below:

Chrome: 87.0.4280.88
Firefox: 83.0

cyber-shifter, 2020-12-15:
So just as further troubleshooting, I tried this is a Chrome browser on another computer that is a Windows computer. I also tried it with Chromium Edge on the same computer. The versions are below:

Edge: Version 87.0.664.60
Chrome: Version 87.0.4280.88

A further development is if I switch the mode to timer, and then back to sensor, the timer settings bars go grey. However, if I hit submit, it does not save. The bar at the top does not appear as it normally does, and the configuration never saves.

This is Terrarium Pi version 3.9.8 on some brand-new Pi 4's.

cyber-shifter, 2020-12-16:
I tried removing the switch in the configuration, and then adding it back again. This did not make a difference. The config still will not save with the sensor option even though the timer fields are greyed out.

theyosh, 2020-12-16:
> So just as further troubleshooting, I tried this is a Chrome browser on another computer that is a Windows computer. I also tried it with Chromium Edge on the same computer. The versions are below: [...]

When you toggle to sensor, and the timer boxes do get grey, will they also have a time set? Because, the idea is that if you use sensor, the values for the timer do not matter. But for the browser, they needed to be filled out.

So, maybe this could solve it for you. Put it on timer, enter some values (start and stop times). Do not enter the same twice. Please let there be an hour in between.

Then you switch back to sensor. The fields you just entered, should become grey, but should still have your entered time values. Then you should be able to save it, with a sensor setup.

cyber-shifter, 2020-12-16:
> So just as further troubleshooting, I tried this is a Chrome browser on another computer that is a Windows computer. I also tried it with Chromium Edge on the same computer. The versions are below: [...]

I tried that here. It won't save the config changes. I didn't state if but a couple times I switched the source to "weather" which caused times to appear in the timer fields, then I switched it back and the fields went grey with the time still in there. I have an older version of Terrarium PI on a test Pi, I'm going to check it when I get off work.

cyber-shifter, 2020-12-18:
So I did this on that test system, and it was able to work. I usually try not to do a format/reinstall wantonly, but I'm going to try that here, since this is an odd problem. I will post my results.

cyber-shifter, 2020-12-19:
A reinstall did not make a difference. Are there some other logs I can check? When I look in the system log I have no warnings.

cyber-shifter, 2020-12-19:
I will also note, when I do a reformat of the card, and reinstall Rasbian, and then update Rasbian before I install Terrarium PI.

cyber-shifter, 2020-12-19:
Brought this to a different raspberry pi completely, did a fresh install of everything. Same problem. Tried all kinds of iterations with putting data in the fields, then turning it back to sensor, etc, etc. None of it worked. Is it possible there is something in the newest Pi OS that is conflicting here?

cyber-shifter, 2020-12-19:
I've tried this with three different Pi's and two different sensors (different types to be sure). The temp sensors get a reading, the switches work fine. I just can't get this part of the config to save on any machine. This is the last part that is holding up everything. I can't put it on a timer either because it's not a time-based activation, I need it to be sensitive to temperature to keep it at a certain point.

cyber-shifter, 2020-12-20:
[image]

cyber-shifter, 2020-12-20:
This is what it looked like after I made the mode timer first, then sensor, and tried to save it. It wouldn't save. I clicked the button and it did nothing. This was done on a different computer than my normal one. An up to date Win 10 device, with all current updates installed including Chrome.

cyber-shifter, 2020-12-20:
I think I made a bit of progress. I went through LOTS of iterations. I had read in another issue post about how if we're doing a temperature sensor setup for the environment, we didn't need to configure the high alarm, to just use the low. So I hadn't been touching the high alarm config. However, I wanted to try it today, since I seemed to be getting nowhere. So I did the following:

- Set mode to sensor.
- Set up low alarm 
- When you select a power switch for the low alarm, the timer fields open
- I put values in the timer fields as well, then attempted to save
- When that didn't work, I set up the high alarm as well, and attempted to save
- This time it worked! but I didn't want the high alarm, so I took out the switch from the high alarm config
- I saved again, and it saved and started functioning properly. I also noticed that the asterisks on the high alarm config went away after I removed the power switch from the bottom option.

I am attempting to test this theory with another Pi right now. Will report back.

theyosh, 2020-12-21:
Hi, a lot of info. But I think I found at least one issue with your step 3 above. I also see when I change the power switches the timer fields get active. That is not right. And this could be the trigger for other strange issues on the same page. I will look into this

cyber-shifter, 2020-12-21:
Thanks! Sorry to dump so much information, but I know when iterating things, it's best to collect data. Since I don't have all my environments running, I have spare parts to test a bunch. I just wanted to give what I was seeing in my testing. I will try to make it more organized in the future. 😅

theyosh, 2020-12-21:
Made a small change. Do a `git pull` and restart. Make sure you clear your browser cache. This should fix the 'step 3' of your step list.

But I am always able to save the page. When you are not able to save the page, I guess there is an error. Look at the screen session to see if there are issues after you try to save. Or, look in your browser debug console for javascript errors.

cyber-shifter, 2020-12-21:
Roger that. I will look in both and try again this evening after work. I glanced in the log file, and even turned the debug logging on, but didn't see anything. But admittedly forgot to check the screen session.

cyber-shifter, 2020-12-22:
Ok, the timer fields do not become active when I select a power switch. Excellent! I'm going to do a full wipe and reinstall on another Pi, and make sure it does not do the same thing. The only error I saw when I loaded the page for the first time was this:
`DevTools failed to load SourceMap: Could not load content for http://172.23.5.21:8090/static/js/hls.min.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE`

theyosh, 2020-12-23:
That error in the error console is not an issue. It wants a .js.map file for debugging but is not available. That is all.

cyber-shifter, 2020-12-23:
Excellent. I thought that was the case. I haven't done that test yet, busy past couple days. I'll try to get to it tonight or tomorrow.

theyosh, 2021-01-04:
Any updates on this? I think it should be fixed with the fix of 2 week ago

cyber-shifter, 2021-01-04:
Yeah, the timer fields don't turn on now. I need to update my other installs, I've just had the holidays keeping me busy. I'll close this now. I know where to go if I have more problems. 😜

closed 2021-01-04 by cyber-shifter