theyosh/TerrariumPI #460: Advise needed:  Humidity Sensors

darrynl, 2020-09-25:
Hi All,

Which humidty sensor works well (ie. is accurate) with TerrariumPI?

Thank you.

bggangel, 2020-09-27:
very bad experiences with DHT22.
Better experiences with 2 BME280's from Waveshare. (from amazon.com) On a pi3B+ it would give me an error after about 24 hours that would only be fixed with a power off/power on cycle, but it might have been my pi3.
I recentl replaced the pi3B+ with a pi4 and I have had no probems in about a week.

theyosh, 2020-10-05:
I would go with BME280 sensors. And if you need to use it in a high humidity environment or using a spray system, put them in a small plastic enclosure and make some holes in that. So the sensor will not be hit by direct water spray.

The-Madhouse, 2020-10-07:
about the DHT22 sensors, i have 2 in my terrarium, and they are realy picky. once every X hours 1 of them would go in an error. and then a reboot was al that could fix that. error's are random, or they just give output NULL, or they just seem to work and still give an error. i just tried something that seems to work, 2 days ago i painted the housings of them and noticed verry bad soldering on the litle pcb's my sensors are connected to so i just put my soldering iron on the highest setting and just touched all 6 pins so the solder woud just reflow. and now we are 2 days later and ( knock on wood!) they have not given any kind of errors. is this the solution for this inexspencive sensors to be used more often? because they are realy handy, temp and humidity in 1 small sensor housing,

theyosh, 2020-10-10:
Hmm, in my opinion, the BME280 is same size as a DHT but does more. BME280 does have temp en humidity but also air pressure and with some math, the altitude. So for me, BME280 always wins over DHTXX.

darrynl, 2020-10-15:
Thanks all for the info.

I have found these:
https://neotronics.co.za/bme280-in-sa
https://www.robotics.org.za/AF2652

that are available locally.
The difference in price and design is quite big.
Which one would you think would be the best for TerrariumPI?

TIA

Darryn

theyosh, 2020-10-21:
Pick the cheapest one.... buying a more expensive one can always be done if really needed...;) 
But I think the cheapest one will just work.

theyosh, 2020-10-30:
Any updates? Can we close this issue?

theyosh, 2020-11-27:
Closing due to no activity

closed 2020-11-27 by theyosh