theyosh/TerrariumPI #463: [BUG]:

martynwheeler, 2020-10-03:
- Python version: [3]
 - Raspberry PI: [3B]

**Describe the bug**
Having just upgraded using git pull and update the motd no longer appears in the console when I log in.  Everything seems to be working okay.

Thanks in advance,

Martyn

theyosh, 2020-10-05:
Is this still the case? I can take some time (1:30) before the first MOTD is generated.

And there are no errors in the screen session: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

martynwheeler, 2020-10-05:
Thanks,

I will check later when I get home (I can't access from work).

martynwheeler, 2020-10-06:
No, it is still missing.  There are some errors in the screen session, does this help??

</pre>
<h2>Traceback:</h2>
<pre>
Traceback (most recent call last):
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 960, in wsgi
    out = self._cast(self._handle(environ))
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 913, in _cast
    out = self.error_handler.get(out.status_code, self.default_error_handler)(out)
  File &quot;/home/pi/TerrariumPI/terrariumWebserver.py&quot;, line 512, in error404
    return template(&#039;404&#039;,**variables)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3630, in template
    return TEMPLATES[tplid].render(kwargs)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3419, in render
    self.execute(stdout, env)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3406, in execute
    eval(self.co, env)
  File &quot;/home/pi/TerrariumPI/views/404.tpl&quot;, line 12, in &lt;module&gt;
    % include(&#039;inc/page_footer.tpl&#039;)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3396, in _include
    return self.cache[_name].execute(env[&#039;_stdout&#039;], env)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3406, in execute
    eval(self.co, env)
  File &quot;/home/pi/TerrariumPI/views/inc/page_footer.tpl&quot;, line 2, in &lt;module&gt;
    document.title = &#039;{{page_title}} | {{title}} ({{version}})&#039;;
NameError: name &#039;version&#039; is not defined

</pre>
<h1>Critical error while processing request: /webcam//a94df035fc91a158c1f9d7a1c769ca66/stream.m3u8</h1><h2>Error:</h2>
<pre>
NameError(&quot;name &#039;version&#039; is not defined&quot;)
</pre>
<h2>Traceback:</h2>
<pre>
Traceback (most recent call last):
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 960, in wsgi
    out = self._cast(self._handle(environ))
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 913, in _cast
    out = self.error_handler.get(out.status_code, self.default_error_handler)(out)
  File &quot;/home/pi/TerrariumPI/terrariumWebserver.py&quot;, line 512, in error404
    return template(&#039;404&#039;,**variables)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3630, in template
    return TEMPLATES[tplid].render(kwargs)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3419, in render
    self.execute(stdout, env)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3406, in execute
    eval(self.co, env)
  File &quot;/home/pi/TerrariumPI/views/404.tpl&quot;, line 12, in &lt;module&gt;
    % include(&#039;inc/page_footer.tpl&#039;)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3396, in _include
    return self.cache[_name].execute(env[&#039;_stdout&#039;], env)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3406, in execute
    eval(self.co, env)
  File &quot;/home/pi/TerrariumPI/views/inc/page_footer.tpl&quot;, line 2, in &lt;module&gt;
    document.title = &#039;{{page_title}} | {{title}} ({{version}})&#039;;
NameError: name &#039;version&#039; is not defined

</pre>

theyosh, 2020-10-06:
This is a strange error. It looks like you settings.cfg file is missing the version setting in the [terrariumpi] section.

Could you post the first 20 lines of settings.cfg? Make sure you do not past passwords

martynwheeler, 2020-10-06:
Hi, 

Here it is:

[terrariumpi]
host = ::
port = 8090
version = 3.9.8
title = TerrariumPI %(version)s
power_usage = 5
owfs_port = 4304
admin = ***********
password = **************
language = en_US
temperature_indicator = C
distance_indicator = cm
power_price = 0
water_price = 0
soundcard = bcm2835 ALSA
always_authenticate = false
windspeed_indicator = kmh
volume_indicator = L
sensor_gauge_overview = false
hide_environment_on_dashboard = false
graph_smooth_value = 0
external_calendar_url =
graph_show_min_max_gauge = true
horizontal_graph_legend = true
meross_password =
meross_username =

theyosh, 2020-10-06:
Have you restarted it already? Because I have no clue at the moment....

martynwheeler, 2020-10-06:
I have restarted it and still the same.  All I did was to follow the instructions on updating the installation.  I also note that the version is not displayed on the webpage?  All the sensors and timers are working though.

[image]

Any suggestions would be welcomed?
Thanks
Martyn

theyosh, 2020-10-06:
Ok, then for now, edit the settings.cfg and remove the %(version)s part from the title setting and restart... if that does not solve it... then I need a good night sleep to figure this out..

martynwheeler, 2020-10-08:
Thanks, that has fixed the motd.  Still getting the error in screen -r

you can see the name on the tab is not quite right either?

[image]

`NameError(&quot;name &#039;version&#039; is not defined&quot;)
</pre>
<h2>Traceback:</h2>
<pre>
Traceback (most recent call last):
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 960, in wsgi
    out = self._cast(self._handle(environ))
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 913, in _cast
    out = self.error_handler.get(out.status_code, self.default_error_handler)(out)
  File &quot;/home/pi/TerrariumPI/terrariumWebserver.py&quot;, line 512, in error404
    return template(&#039;404&#039;,**variables)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3630, in template
    return TEMPLATES[tplid].render(kwargs)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3419, in render
    self.execute(stdout, env)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3406, in execute
    eval(self.co, env)
  File &quot;/home/pi/TerrariumPI/views/404.tpl&quot;, line 12, in &lt;module&gt;
    % include(&#039;inc/page_footer.tpl&#039;)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3396, in _include
    return self.cache[_name].execute(env[&#039;_stdout&#039;], env)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3406, in execute
    eval(self.co, env)
  File &quot;/home/pi/TerrariumPI/views/inc/page_footer.tpl&quot;, line 2, in &lt;module&gt;
    document.title = &#039;{{page_title}} | {{title}} ({{version}})&#039;;
NameError: name &#039;version&#039; is not defined

</pre>
<h1>Critical error while processing request: /webcam//a94df035fc91a158c1f9d7a1c769ca66/stream.m3u8</h1><h2>Error:</h2>
<pre>
NameError(&quot;name &#039;version&#039; is not defined&quot;)
</pre>
<h2>Traceback:</h2>
<pre>
Traceback (most recent call last):
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 960, in wsgi
    out = self._cast(self._handle(environ))
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 913, in _cast
    out = self.error_handler.get(out.status_code, self.default_error_handler)(out)
  File &quot;/home/pi/TerrariumPI/terrariumWebserver.py&quot;, line 512, in error404
    return template(&#039;404&#039;,**variables)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3630, in template
    return TEMPLATES[tplid].render(kwargs)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3419, in render
    self.execute(stdout, env)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3406, in execute
    eval(self.co, env)
  File &quot;/home/pi/TerrariumPI/views/404.tpl&quot;, line 12, in &lt;module&gt;
    % include(&#039;inc/page_footer.tpl&#039;)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3396, in _include
    return self.cache[_name].execute(env[&#039;_stdout&#039;], env)
  File &quot;/usr/local/lib/python3.7/dist-packages/bottle.py&quot;, line 3406, in execute
    eval(self.co, env)
  File &quot;/home/pi/TerrariumPI/views/inc/page_footer.tpl&quot;, line 2, in &lt;module&gt;
    document.title = &#039;{{page_title}} | {{title}} ({{version}})&#039;;
NameError: name &#039;version&#039; is not defined

</pre>
`

martynwheeler, 2020-10-09:
Do you think that it would be worth running the update again?  I'd rather not reinstall as I have everything set up.  However, if there is a way to reinstall and keep all my configuration that would be okay.

Thanks

theyosh, 2020-10-09:
Hmm, you could always do a `git pull` and rerun the installer. In case of a reinstall: https://github.com/theyosh/TerrariumPI/wiki/FAQ#which-files-to-backup

martynwheeler, 2020-10-09:
I will give it a try

martynwheeler, 2020-10-09:
Hmm, an update has not solved it.  Okay, so if I want to reinstall, I save the files you pointed to and remover the Terrariumpi directory and then reinstall by following the installation procedure?  Just want to clarify before I start.

Thanks again for your help

theyosh, 2020-10-09:
Do not remove, but just rename the older directory ;) Makes it easier to copy the files over

martynwheeler, 2020-10-09:
I don't know if this is related in any way but I am also seeing this error:

● owserver.service - Backend server for 1-wire control
   Loaded: loaded (/lib/systemd/system/owserver.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-10-09 19:33:10 BST; 45s ago
     Docs: man:owserver(1)
  Process: 481 ExecStart=/usr/bin/owserver -c /etc/owfs.conf (code=exited, status=1/FAILURE)
 Main PID: 481 (code=exited, status=1/FAILURE)

Oct 09 19:33:10 terrariumpi systemd[1]: owserver.service: Service RestartSec=100ms expired, scheduling restart.
Oct 09 19:33:10 terrariumpi systemd[1]: owserver.service: Scheduled restart job, restart counter is at 5.
Oct 09 19:33:10 terrariumpi systemd[1]: Stopped Backend server for 1-wire control.
Oct 09 19:33:10 terrariumpi systemd[1]: owserver.service: Start request repeated too quickly.
Oct 09 19:33:10 terrariumpi systemd[1]: owserver.service: Failed with result 'exit-code'.
Oct 09 19:33:10 terrariumpi systemd[1]: Failed to start Backend server for 1-wire control.

theyosh, 2020-10-09:
No this is not related. But this can be solved by editing:

edit `/etc/owfs.conf` and enable the line `server: FAKE = DS2405` around line 16. My installed does remove this, but apparently this does not work with the newer versions.

martynwheeler, 2020-10-09:
Mine says,
#server: FAKE = DS18S20,DS2405
Should I remove the DS18S20 as I am using two DS18S20 sensors in my setup

theyosh, 2020-10-09:
No, as this is a fake sensor, it will just generate random values. Only the DS2405 is needed. This will make OWFS work again, and fix the systemd status issue. So remove DS18S20 and remove the '#' in front of the line.

Then test it with service owfs restart and that should then work

martynwheeler, 2020-10-09:
Okay, I have done a new installation and I seem to have got rid of the error.  Thank you for all of your help!

closed 2020-10-09 by martynwheeler

theyosh, 2020-10-09:
Nice and enjoy the new version!

theyosh, 2020-10-10:
He, I found the bug. It has to do when you zoom in with the webcam or loading the favicon.ico file. That give a 404 error, which was missing the version number. So it should now be fixed for ever!

commit ea32bcdd (2020-10-10): Fix missing variable in the 404 page. Fixes #463