theyosh/TerrariumPI #465: [FEATURE]: Raspberry pi Hat

se7en9057, 2020-10-10:
I'm planning on working on creating a RPi hat that will work with this project to make a cleaner setup and easier for people who have little to no tinkering/hacking skills/experience. @theyosh i'd love your blessing, input, support, ideas. would like to start a Git for this and begin the design phase soon. 

requirements of hat:

- active cooling for pi
- easy sensor connectors
- access to GPIO
- LCD / OLED ?
- Custom Rasbin OS install to do config and script for LCD info (IP address etc)

lets start the discussion!

se7en9057, 2020-10-11:
[image]

trying to decide on pinouts and numbers of connectors.  i think the JST-XH connector is universal enough to be useful. still up to suggestions

se7en9057, 2020-10-12:
So far here's what i'm thinking. afew easy access sensor connectors, GPIO connectors that will support random things as well as the relay boards that don't use i2c and some i2c connectors for easy connecting them. not to mention 5v headders along with a 5v active cooling fan. 
[image]
[image]

theyosh, 2020-10-12:
Hmm this is funky. i do not have any experience with making boards, but I do have some idea's which could be totally different than what you have now.

1. Make the connections human readable/usable. EX: https://pinout.xyz/pinout/1_wire is the 1Wire connector. Together with a 3.3v power and a ground, you can hook up a lot of 1 wire sensors. It also needs a resistor to work. So it would be nice the you have 3 pins, where you can hook the 1Wire sensors. So that a person does not have to handle the right resistor. Not sure if this can be done, but there are more boards: https://pinout.xyz/boards

2. And I do use a lot I2C, so I am looking for an easy way to add I2C sensors. So what if, you have 4 times a 4 pins connector that are all connected to the same I2C bus. Then I can just hook up a few sensors without hassling with combining lose wires together. The second image does have 2 connectors, but I think I need more.

3. Then a tricky part. I use I2C multiplexing which means that I have multiple I2C buses. I do this, because some I2C sensors have fixed addresses so you can only have one per I2C bus. This does involve some config changes in the Pi boot-up. Then my wish would be that I could enable this with some kind of jumper, so that a second row of I2C connectors are available. If the jumper is off, they should return to normal state. But this will be hard and a challenge.

4. The fan is rather big. I am currently using a case with a 3cm x 3cm size fan. So with a smaller fan, you have more space for connectors. https://pinout.xyz/pinout/fan_shim

5. Status led indicator: https://github.com/theyosh/TerrariumPI/wiki/Hardware#notification (I assume 3 GPIO pins (colors) to put on or off)

This is what I can come up with.....

se7en9057, 2020-10-13:
@theyosh i'll see what i can come up with :) 
glad to have your input. your software is keeping my carnivorous plants super happy and flourishing! thanks again for all your hard work and effort!

se7en9057, 2020-10-13:
[image]
added more i2C but i still have to research how the i2c multiplexing works. added the one wire sensor ports. 
changed the 40mm fan to 30mm.

theyosh, 2020-10-30:
Something else I thought of. How about make it stackable. So first you have a PCB with a FAN and some basic GPIO pins. The connectors need to be in a 90degree angle, so you can plug it from the side and not from the top.

Then when you want I2C, you can add another PCB (stack it on top) and then you can add some I2C sensors, also with 90degree angle connectors.

So, you have less connectors per PCB, as not everybody needs everything

se7en9057, 2020-11-01:
@theyosh thats a good idea I also have a friend working on some stuff based on this as well. Also do you have a patrieon or something so that if we are able we can contribute to this project monetarily or some other way?

closed 2021-01-08 by se7en9057