theyosh/TerrariumPI #467: [FEATURE]: Color conversion codes for webcam

npinter, 2020-10-13:
I'm using Pi Camera IR-Cut (Sensor OV5647), connected via the RPiCam port. 
[image]

Location is `/dev/video0`. I had some strange color conversion issues (color inversion, pinkish colors..) until I removed the color conversion part from the code.

**642:terrariumWebcam.py**
`jpeg = Image.fromarray(cv2.cvtColor(image,cv2.COLOR_BGR2RGB))`
to
`jpeg = Image.fromarray(image)`

Feature: Adding a dropdown to webcam settings with the choice of activate/deactivate color conversion.

theyosh, 2020-10-13:
Hmm, I am also using such a camera. For me, the color is fixed by using the 'grey world' option in the webcam settings. This will produce nice pictures. Have you tried that?

Because I am not sure what the impact will be for other Pi camera's.

npinter, 2020-10-13:
"greywold" in white balance? This does not fix the issue in my case:

[image]

Removing the color conversion part of the code:

[image]

theyosh, 2020-10-14:
Ok, then you are not using the camera as an RPI cam? Because the color change only happens on a USB camera. Not a RPI camera.

So what is the address you enter for your camera?

npinter, 2020-10-14:
I tried that before, but the camera will not be recognized anymore:

[image]

It only works with `/dev/video0`.

npinter, 2020-10-15:
*facepalm*.. entered the `rpicam` and did the restart as you suggested:

[image]

I guess this can be closed now. Thanks for all the effort you've put into TerrariumPi, it's really nice :)

closed 2020-10-18 by npinter