theyosh/TerrariumPI #468: [FEATURE]: Add switches before and after taking an image with the webcam

npinter, 2020-10-13:
I'm using Pi Camera IR-Cut (Sensor OV5647), connected via the RPiCam port. 
[image]

Only direct sunlight reduces resistance enough of the light dependent resistors to deactivate the LEDs. Due to that, both LEDs stay always on and really heating up (which is also a waste of energy). 

I've removed both LDRs and added a NPN transistor (BC337-40) as well as a 1k Ohm resistor to control both LEDs via GPIO. 
[image]
[image]

**Feature request:** The possibility to set switches before and after taking a picture with the webcam (f.e. IR LEDs on, take picture, IR LEDs off). I'm going to hard code it, but it would be nice if one is able to setup this simply in the UI!

theyosh, 2020-10-30:
Hi, sorry for the late reply, but this is a great hack! I use this one also, and noticed the heat also... But this not bother to much.

So, I added this option as a setting. You can now add ',[GPIO PIN]' to the address: rpicam. So then you get the final address like: rpicam,22 where 22 is the physical pin number. https://github.com/theyosh/TerrariumPI/wiki/Hardware#gpio-power-saving

But, can I ask you then to make a small schema (with fritzing) how the hardware is hooked up? So what kind of resistor. So that I can add that to the power management wiki page. So other people can also try this.

I used a different approach. But the result should be the same. Could/would like to test it?

theyosh, 2020-11-08:
Hi, did you have time to test this solution? And do you have time to make a small schema of the resistor and wiring it to the Pi?

npinter, 2020-11-08:
Wow nice thank you! I have to test this! Sorry for the late reply, too ;)

theyosh, 2020-12-03:
Hi, did you had time to make a small schema? If it is easier, just make a small drawing, and I create the schema in fritzing?

I really would like to add this hack to the Wiki.

npinter, 2020-12-23:
Sorry!! I'm pretty busy right now, but I'll have some spare time in the next few days to finally test it and create some kind of soldering scheme :)

npinter, 2021-01-11:
Sorry for the delay (again).. here is the scheme. fritzing was to complicated so I just used my pictures from above!

[image]

npinter, 2021-01-11:
Sorry for the delay (again).. here is the scheme. fritzing was too complicated so I just used my pictures from above!
[image]

closed 2021-02-19 by theyosh