theyosh/TerrariumPI #469: Failed to start terriumpi.service: Unit terriumpi.service not found

vast321, 2020-10-18:
pi 3：
pi@raspberrypi:~/TerrariumPI $ sudo service terriumpi start
Failed to start terriumpi.service: Unit terriumpi.service not found

Browser opens Web address：
A connection to the 192.168.0.105:8090 server could not be established.

Matze75-Swiss, 2020-10-18:
May be the was not completely installed. 
Please check "Interface Can't Be reached #452" under "closed issues"

theyosh, 2020-10-18:
Manual starting: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-start-terrariumpi

Systemd is not used. If you need/want that: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-systemd

vast321, 2020-10-19:
@Matze75-Swiss  Thank you!

vast321, 2020-10-19:
@theyosh   Thank you for your reply!

pi@raspberrypi:~/TerrariumPI $ python3 terrariumPI.py
.^HTraceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumAudio.py", line 18, in <module>
    from MediaInfoDLL import MediaInfo, Stream
ModuleNotFoundError: No module named 'MediaInfoDLL'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "terrariumPI.py", line 21, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 35, in <module>
    from terrariumAudio import terrariumAudioPlayer
  File "/home/pi/TerrariumPI/terrariumAudio.py", line 20, in <module>
    from MediaInfoDLL3 import MediaInfo, Stream
ModuleNotFoundError: No module named 'MediaInfoDLL3'

vast321, 2020-10-19:
sudo apt install python3-mediainfodll

theyosh, 2020-10-19:
Hmm, you can always rerun the installer to make sure everything is installed. I advise to use Python3. It could be that something is not fully installed.

`sudo ./install.sh 3`

theyosh, 2020-10-30:
Any other issues? Can we close this issue?

closed 2020-11-11 by vast321