theyosh/TerrariumPI #470: Using DHT11 / DHT11 on Raspberry Pi 4b

Matze75-Swiss, 2020-10-18:
This is not really a bug report, but may be it helps others trying to install TerrariumPi on a Raspberry Pi 4b (Hardware Version BCM2711).

Using DHT22 or DHT11 Sensors will end up in an error like this:
Traceback (most recent call last):
  File "./AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 94, in read_retry
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 80, in read
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 60, in get_platform
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/Beaglebone_Black.py", line 24, in <module>
ImportError: cannot import name Beaglebone_Black_Driver

The reason is, that the Adafruit_DHT Version 1.4.0 not yet knows the new Raspberry Pi 4b CPU (may be they will newer do as the Adafruit_DHT library has been deprecated) and recognizes the Pi 4b as a Beagle Bone ... 

You can easily check what hardware version is used by running: cat /proc/cpuinfo 
In the last four lines you will find the following (see Hardware):
Hardware	: **BCM2711**
Revision	        : d03114
Serial		: 10000000xxxxxxx
Model		: Raspberry Pi 4 Model B Rev 1.4

Solving that problem is quite simple; you just have to extend the file **platform_detect.py** which is located in the folder: /usr/local/lib/python3.7/dist-packages/Adafruit_DHT/ if you are using Python3, otherwise in /usr/local/lib/python2.7/dist-packages/Adafruit_DHT/ 

So move to the folder with: **cd /usr/local/lib/python3.7/dist-packages/Adafruit_DHT/**
and edit the file by commanding: **sudo nano platform_detect.py** 

Now move to the end of the file and insert the following before the **_else:_**:
`elif match.group(1) == 'BCM2711':`
        `# Pi 4b+`
        `return 3`

Alternatively you can change the very last line from:
`return None`
to 
`return 3`

Then save the file and reboot the Pi. 

Now everything should work properly.

I hope this tip helps someone and saves them a day of searching like i did.

theyosh, 2020-10-30:
Hmm, thanks for the info. That is a reason that in the new version, I will not use this library any more. But also, I will push a bit more to BME280 and drop active support for DHTXX sensors. If the DHT works, it works... if not... to bad.

They are nice to play with, but not for running continuously for years.

Matze75-Swiss, 2020-11-01:
Yes, you are absolutely right. Unfortunately the DHT22 was the only sensor I had available when I started the new terrarium.
Off course, the BME280 is much better and much more stable over time; so I planed to change to that sensor when I make major clean in the terra next time. Actually I was clueless when I changed to Raspi4 and I don't want to stress my crocodile directly, thus I made some software research :-)

theyosh, 2020-11-01:
He, I do not have a crocodile in #210 :P

Hakx32, 2020-11-01:
I am new to this and I am trying to get it set up for my Bearded Dragon, I still cant get the DHT22 to work even after adding the info above to the platform_detect.py file. Is there anything else I can do without changing out the sensors for the BME280?

theyosh, 2020-11-01:
Hmm, I am afraid not. As there are a lot of issues with DHT: https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed+DHT I am at the opinion that supporting DHT is not worth the time to debug it all the time.

DHT works nice for some small project that does not operate 24/7. But for 24/7, I only see problems with DHT

Hakx32, 2020-11-01:
Thanks I guess I will look into getting the BME280 sensors

theyosh, 2020-11-01:
And if you buy them, buy 2 or 3. From experience it is handy to have spare parts ;)

Matze75-Swiss, 2020-11-01:
Hmm, the DHTxx are tricky. If you have running all the other stuff and there is no error (library correctly installed and HMI reachable via web server) the only thing i have in mind is the correct addressing of the sensor. Please use the physical pin number, not the GPIO number (e.g. 18 instead of GPIO24).
[image]
Further if you see any but wrong values and using relatively long cables (1m and more) try to decrease the resistor between voltage and sensor pin. Normally it is 4.7 kOhms and soldered on the PCB depending on the type of sensor you use. Try to decrease the resistor to 2 kOhms. That was also an Error I had...

But using the BME280 would be off course the better decision...

victorhooi, 2020-12-28:
Aha, that is interesting - I did not know the DHT22/AM2302 sensors were no longer recommended.

Just saw this article as well, which also discussed it:

https://hackaday.io/project/28775-my-smart-fishroomanimal-room/log/72341-chosing-a-humidity-sensor-skip-dht22-go-for-bme280

@theyosh Is the suggestion for TerrariumPi users to get the Bosch BME280 sensor, as a drop-in replacement for the DHT11/DHT22/AM2302? Would it make sense to update the wiki, to make this clear? (Happy to submit a PR, if that helps, and once I understand things better).

Adafruit is selling the [BMED280 with Stemma QT connectors](https://www.adafruit.com/product/2652). Is that recommended, or is there another vendor you suggest?

Is there some kind of plastic enclosure, or case you'd suggest, assuming you were putting this into a fairly humid (~80%) enclosure?

What about for the waterproof use-case - e.g. DS18B20 is available in a [waterproof version](https://www.adafruit.com/product/381), which is useful for example measuring the water at the bottom of my terrarium. Is there a newer alternative you suggest there?

theyosh, 2020-12-28:
If your DHT is running well... I think you can consider yourself lucky :P . But there a many people complaining that the DHT stops working after a few months. It looks like they are not made to work 24/7/365. And the BME280 are working  nicely. The only issue I have that I was not able to change the address, so only one BME280 per I2C bus.

About the Wiki. It needs to be updated much more. Specially when version 4.0 is available. So for now it is fine. And most people who asked about DHT issues, are told to upgrade. It is not a big issue.

And I use all kind of BME280 sensors from Amazon. https://www.amazon.com/Organizer-Temperature-Humidity-Atmospheric-Barometric/dp/B07V5CL3L8/ I prefer 5Volt versions, as the Pi has more power (amps) on the 5Volt system then the 3.3Volt system. But these sensors needs some soldering with wires. For me not an issue. Done that many times.

I have in my 'wet' terrarium (humidity > 60%) just small plastic cases like: https://www.amazon.com/Abgream-Plastic-Beads-Storage-Containers/dp/B0825PSQ76/ . But basically, just get a water tight case, and make some small holes in it so the air can go through and stick it somewhere. That is enough to measure, but it will not get splashed by water. Also high humidity is not an issue for the hardware, as long it is not salt water. That is causing why more corrosion. And direct water can cause issues.

And for the DS18B20 I would just go for the most cheapest, as long there is a metal head on it. Then they are waterproof: https://www.amazon.com/Gikfun-DS18B20-Temperature-Waterproof-EK1083x3/dp/B012C597T0/
I use those to measure my spraying water temperature. It is just laying on the bottom of my water tank. No problems.

theyosh, 2021-10-15:
Closing as it is now at the hardware page of this type of sensor. https://github.com/theyosh/TerrariumPI/commit/05edbc333699cc9b2a34b82ccf4cc4ce9ec3108d 

and https://theyosh.github.io/TerrariumPI/hardware/sensor/dht11-sensor/

closed 2021-10-15 by theyosh