theyosh/TerrariumPI #471: robotdyn AC dimmer support+push button support

The-Madhouse, 2020-10-20:
dear, 

i'm trying to use a robotdyn pwm ac dimmer i have laying around, but it does not seem to work with terrariumpi (in my head all pwm dimmer are the same...guess i'm realy wrong!!) 
https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html 
tried connecting, but just goes to 100% brightness when connected. 
it also has a 4the pin ( vcc, gnd, ZERO? and pwm singnal) and i can't find how to connect it. 

also is it possible to add support for a push button to activate a relais? i have red lighting in my terrarium and i want to use it as a quick check light if all lights are of, so a push button to activate 1 relais would be sweet. 

thanks for reading

martynwheeler, 2020-10-20:
Not sure there are RPi libraries for this module - looks like it is
designed for arduino. Might be worth digging around RPi forums to see if
anyone has ported it to the Pi.

The-Madhouse, 2020-10-20:
i have just send an email to robodyn with the question for libraries and maybe a litle more info about the 4the pin. 
if they reply this could be usefull for some people here, because the modules are realy niceand sturdy looking.

AndrejWeber, 2020-10-20:
> i have just send an email to robodyn with the question for libraries and maybe a litle more info about the 4the pin. 
> 
> if they reply this could be usefull for some people here, because the modules are realy niceand sturdy looking. 
> 
> 
> 
> thanks for reply ;-) 

The module will not work on raspi, it needs a detection of zero crossing of AC-sinus-wave.
There are also hardware, which can detect zero crossing, but then you need the right output signal, to trigger your cheap dimmer. 
It will be much more expensive to test and find the right one, than buy a dimmer, which is supported by terrariumPi. 
There are also more requests here to make it possible by TheYosh, but the big problem is, that a raspberry pi is much to slow to detect and trigger in real time.
A Arduino is able to manage this detection of zero crossing. But someone has to write a program for the arduino.

The-Madhouse, 2020-10-20:
ok, the magical words missing in all my google searching where " zero crossing" , 
after searching with these words on the dimmer for raspberry i indeed found that the only way to get this to work is to use an arduino in the mix... and write some code for it. ok well...thanks, if anybody from belgium needs a perfectly good AC dimmer, that works perfectly on an arduino let me know :-p  

and the adding of the push buttons is also this difficult? i asume not because you don't need zero crossing detection :-) :-p 

thanks for reply's people

theyosh, 2020-10-30:
And once again @AndrejWeber is here. :) Thanks for the explanation. Somehow, you always shows up when dimmer questions.

But the only AC dimmer known to work is: https://github.com/theyosh/TerrariumPI/wiki/Hardware#nextevo It is not cheap, specially shipping. But if you buy a few (spare) then it is worth the money. I have used now one for more then 4 years without any problems.

The-Madhouse, 2020-10-30:
Ik found a different solution, a module you can place in between your lights and switch, it is used a lot with birds. Does sunrise and sunset perfectly.(you can even set thé dimming time between 2seconds and 2hours) It's made by a dutch Guy and costs around 35€.... Bought 2 , one led one and one normal one. It's a lot of money, but it's more then worth it imho, i'm really happy when i look at my terrarium and they sun goes up or down...

theyosh, 2020-10-30:
Have a link?

theyosh, 2020-11-08:
@The-Madhouse Do you have some more info like a website? I would like to know more

The-Madhouse, 2020-11-08:
https://www.mijnwebwinkel.nl/winkel/dimmers/ 

@theyosh did you receive my virtual coffee? ;-)

theyosh, 2020-11-09:
Yup. Thanks!. And I am in contact with the dimmer shop. So I think we can close this issue?

The-Madhouse, 2020-11-09:
It can be closed when there is An answer on thé question about thé push buttons? And i'm also back in contact with thé seller of the dimmers because mine are not 100% flicker free, seller is verry helpfull.

theyosh, 2020-11-09:
I ordered also one. He is actually Belgian. And I asked for a small change so that I can toggle on/off with the Pi the directly. Lets see if that works.

For the button, I think this is something you have to make yourself. The issue is is the button a hardware button? I do have an example script that could help: https://github.com/theyosh/TerrariumPI/blob/master/contrib/toggle_switch.py

I think the script is pretty readable. And the trick is to fit it to your needs.

theyosh, 2020-11-27:
Is my example for a custom button enough? Else I do not have a solution. Would like to close this issue

The-Madhouse, 2020-11-28:
You can close this, project is put on a shelf for this time because of all thé issues i have with it and thé (most important!) Lack of skill from my end.did not try thé sample button code.

closed 2020-12-03 by theyosh