theyosh/TerrariumPI #472: Use a buzzer when an alert is on

kebs19, 2020-10-23:
I think it would be nice to have a way to sound a buzzer when the system has an error. I see that there are small buzzer modules that could be driven by a gpio pin... maybe make it sound for a while?

It's quite common for me to have an error asking to turn the system off and back on (it's just a faulty sensor, but it happens like once in a month, so it's fine by me), and it gives me an alert using Pushover, but when I'm not at home... a sound alert would be a good way to tell people around the aquarium that it needs a good reboot. :)

theyosh, 2020-11-17:
Hi,

sorry for not responding earlier. This could be a nice addition. But I have to see how easily I can add it to the notification system. Also, I will only support active buzzers (which makes noise when powered on). There are buzzers that can play tunes, but for that we need PWM. And I use PWM for dimmers.

So, I try to add this this weekend in the notification part.

theyosh, 2020-12-03:
Again, sorry for the slow responding. I have thought this over again, and put it on the wish list for now. As I am working an a complete refactor of the software, where this can be easier integrated. At the moment, the notification system code is a bit of a mess.

So I put this on the wish list, and I expect to implement this in the new 4.X version.

kebs19, 2020-12-03:
Hey there! It's fine, I got excited to know that you're working on a v4 😁😁
What I'm doing right now is using a sonoff smart switch and pushover notifications, so when I see that the system is halting I can turn it off and back on using the sonoff switch.
I thought about using the Pi to reset itself but I'm not able to power it off and then back on, let alone the fact that exposing the system to the Internet is a big no-no to me.

theyosh, 2020-12-03:
Hmm, maybe it can be done easier. There is a watchdog script https://github.com/theyosh/TerrariumPI/blob/master/watchdog.sh that checks if the logfile is being updated. When not, it can trigger something.

And using Sonoff with Tasmota software, you can call: http://[device]/cs?c2=197&c1=Backlog%20Power%20OFF%3B%20Delay%2030%3B%20Power%20ON"

This will restart the Sonoff with a 30 second delay. And this can be fired on the Pi it self. When the request is OK, it will reboot the Sonoff. And in 45 sec you should have a working system again.

Assuming that the Pi itself keeps running.

theyosh, 2022-01-27:
Hi, in version 4.x.y.z there is now a Buzzer service. When you add that, you will see a list of 'songs' that can be played. Copy the exact name (including casing) in the *subject* of the message type. Then it will play that song when the message occurs. There is no need to enter a message

theyosh, 2022-02-13:
Closing due to a new release with this functionality

closed 2022-02-13 by theyosh