theyosh/TerrariumPI #475: TerrariumPI not starting after update

darrynl, 2020-11-01:
Hi All,

After doing an update to TerrariumPI, it no longer starts.
I tried doing an update on my running V3.9.8 installation just to make sure that I had the latest and greatest, but it looks like I broke something ...

What I did:

Logged into Raspberry Pi as user pi
cd TerrariumPI
git pull
sudo ./install.sh 3
When the installer as done I chose to reboot.
When the Pi came up, the web interface was not reachable and the system is not listening on port 8090.

Trying to start TerrariumPI manually:
[image]

theyosh, 2020-11-01:
Hmm, somehow you luma.oled package is outdated... try:

`sudo pip3 install -U luma.oled`

Or, rerun the installer. Maybe some packages did not install correctly...

darrynl, 2020-11-01:
Thank you.

[image]

I already tried re-installing and got the same result. :-)

theyosh, 2020-11-01:
And it is running again?

darrynl, 2020-11-01:
No,  it is still not running.

theyosh, 2020-11-01:
But with the same error still? Because when I open the file: /usr/local/lib/python3.7/dist-packages/luma/core/bitmap_font.py there is no reference to import UnidentifiedImageError. 

```
# -*- coding: utf-8 -*-
# Copyright (c) 2020 Richard Hull and contributors
# See LICENSE.rst for details.

from pathlib import Path
from math import ceil
from copy import deepcopy
from PIL import Image, ImageFont
import cbor2
from luma.core.util import from_16_to_8, from_8_to_16
```

And I see that I have a higher luma.core version. So could it be that running `sudo pip3 install -U luma.core` will save this? Then it looks like a dependency issue with the luma modules...

darrynl, 2020-11-02:
I ran a sudo pip3 install -U luma.core this morning and now it seems to be working.

Thank you for your help.

closed 2020-11-02 by darrynl