theyosh/TerrariumPI #476: [BUG]: Engine loop: Sensor has problems

5point0, 2020-11-03:
Hi @theyosh 

I setup some alerts and enabled system error notification. Since then I have been seeing this error come up but seems to stop after a while. Is this something I should look into?

**Setup:**
 - Python version: 3
 - Raspberry PI: 4

TypeError: '<=' not supported between instances of 'float' and 'NoneType'
if current is None or not (self.get_limit_min() <= terrariumUtils.conver_to_value(current,self.get_indicator()) <= self.get_limit_max()):
File "/home/pi/TerrariumPI/terrariumSensor.py", line 115, in update
self.sensors[sensorid].update()
File "/home/pi/TerrariumPI/terrariumEngine.py", line 614, in __engine_loop
Traceback (most recent call last):
2020-11-03 12:08:48,512 - ERROR   - terrariumEngine      - Engine loop: Sensor has problems: '<=' not supported between instances of 'float' and 'NoneType'
TypeError: '<=' not supported between instances of 'float' and 'NoneType'
if current is None or not (self.get_limit_min() <= terrariumUtils.conver_to_value(current,self.get_indicator()) <= self.get_limit_max()):
File "/home/pi/TerrariumPI/terrariumSensor.py", line 115, in update
self.sensors[sensorid].update()
File "/home/pi/TerrariumPI/terrariumEngine.py", line 614, in __engine_loop
Traceback (most recent call last):
2020-11-03 12:08:48,262 - ERROR   - terrariumEngine      - Engine loop: Sensor has problems: '<=' not supported between instances of 'float' and 'NoneType'

theyosh, 2020-11-03:
Could you try a `git pull` first? Based on the linenumbers in the error, I see different code. That could indicate you have not the latest version of the code.

And could you also upload the terrariumpi.log file? There it should show a line that it is updating a sensor. The last update line before the error, should contain the sensor that is causing this problem.

5point0, 2020-11-03:
I am running older code because everything has been running smoothly and i'm worried if I update I break something.😅

I have uploaded the log file, maybe its something I can fix without updating?
[terrariumpi-5.log](https://github.com/theyosh/TerrariumPI/files/5480770/terrariumpi-5.log)

theyosh, 2020-11-03:
Hmm, strange. It only happened a few times, and it is gone. So that is strange. I can imagine, that is has happened when you saved the new notification settings. Then it should not return as long as you do not change the notification settings.

If you have now setup the right settings, I think for now, you can restart once, and see if it comes back. If not, I call this a hickup

5point0, 2020-11-04:
So it came back after restart.
At about 06:15am on the log, but after a few of that error I got an error loading W1 sensor so I'm not sure if its related but they both stopped by themselves.

[terrariumpi-6.log](https://github.com/theyosh/TerrariumPI/files/5485882/terrariumpi-6.log)

theyosh, 2020-11-04:
It looks that the 1wire is somehow the cause of this... Because when you look in the log around 06:14 you will see:

```
2020-11-04 06:14:39,168 - INFO    - terrariumSensor      - Updated mitemp sensor 'Auto Temp' temperature from 27.80C to 27.80C in 0.00016 seconds
2020-11-04 06:14:39,276 - INFO    - terrariumSensor      - Updated mitemp sensor 'Auto Hum' humidity from 47.40% to 47.40% in 0.00014 seconds
2020-11-04 06:14:39,387 - INFO    - terrariumSensor      - Updated w1 sensor 'Prob 2' temperature from 23.88C to 23.88C in 0.00014 seconds
2020-11-04 06:14:39,497 - INFO    - terrariumSensor      - Updated w1 sensor 'Prob 1' temperature from 21.94C to 21.94C in 0.00014 seconds
2020-11-04 06:14:39,606 - INFO    - terrariumSensor      - Updated k30co2 sensor 'CO2' co2 from 490.00ppm to 490.00ppm in 0.00015 seconds
2020-11-04 06:14:39,716 - INFO    - terrariumSensor      - Updated bme280 sensor 'Temperature BME' temperature from 23.52C to 23.52C in 0.00015 seconds
2020-11-04 06:14:39,826 - INFO    - terrariumSensor      - Updated bme280 sensor 'Humidity BME' humidity from 51.89% to 51.89% in 0.00014 seconds
2020-11-04 06:14:39,936 - INFO    - terrariumSensor      - Updated bme280 sensor 'Test light' temperature from 24.11C to 24.11C in 0.00013 seconds
2020-11-04 06:14:40,095 - INFO    - terrariumSensor      - Updated bme280 sensor 'Hum light' humidity from 52.74% to 52.74% in 0.00013 seconds
```

After the mitemp sensors, there are two 1 wire sensors. And the next round, you get the error after the mitemp, and the w1 sensor are not on the log anymore. So it does look like the 1wire is causing this. Not sure, why the actual 1 wire error is later.

but at 06:20 it looks all ok again:
```
2020-11-04 06:20:39,423 - INFO    - terrariumSensor      - Updated mitemp sensor 'Auto Temp' temperature from 27.80C to 27.80C in 0.00016 seconds
2020-11-04 06:20:39,531 - INFO    - terrariumSensor      - Updated mitemp sensor 'Auto Hum' humidity from 47.50% to 47.50% in 0.00011 seconds
2020-11-04 06:20:40,485 - INFO    - terrariumSensor      - Updated w1 sensor 'Prob 2' temperature from 23.88C to 24.00C in 0.84372 seconds
2020-11-04 06:20:41,446 - INFO    - terrariumSensor      - Updated w1 sensor 'Prob 1' temperature from 21.94C to 22.12C in 0.84934 seconds
2020-11-04 06:20:41,556 - INFO    - terrariumSensor      - Updated k30co2 sensor 'CO2' co2 from 486.00ppm to 486.00ppm in 0.00018 seconds
2020-11-04 06:20:41,673 - INFO    - terrariumSensor      - Updated bme280 sensor 'Temperature BME' temperature from 23.55C to 23.55C in 0.00016 seconds
2020-11-04 06:20:41,785 - INFO    - terrariumSensor      - Updated bme280 sensor 'Humidity BME' humidity from 51.71% to 51.71% in 0.00014 seconds
2020-11-04 06:20:41,895 - INFO    - terrariumSensor      - Updated bme280 sensor 'Test light' temperature from 24.16C to 24.16C in 0.00014 seconds
2020-11-04 06:20:42,005 - INFO    - terrariumSensor      - Updated bme280 sensor 'Hum light' humidity from 53.24% to 53.24% in 0.00014 seconds
```

So, could it be that there is a cronjob active around 06:14 or 06:15 ? Because it could be that the Pi is busy with something which makes the 1wire sensors fail.

But according to the log, everything will continue to work, and I expect a small flat line in the 1 wire graphs

5point0, 2020-11-06:
Okay, so this has probably always been happening but I only saw it now because I enabled the notifications and see the errors as they happen when before it would go away so I didn't see it.

The only cronjob I have running is to update the dns every 5 min for my VPN.
I checked and secured the 1 wire sensors wiring.

Will keep an eye and see what it does.

theyosh, 2020-11-06:
In the log directory there is a log history of 30 days. Maybe you can find there when it started

5point0, 2020-11-06:
I have checked the log and seems its been happening all along. I can find the error on days before I changed notification settings. It also seems to be random, every other day and different times.

I think I must be related to the 1 wire sensors as they also have been giving an error every now and then as far back as the log goes.

Since I have secured the cabling I haven't seen the issue again but if it does come back I will try to rewire the sensors and see if that helps.

theyosh, 2020-11-06:
On the other hand, I made my software to work with cheap sensors (that can fail some times). So that there is a few errors a day, that is not a problem. It can sometimes looks more worse then it really is.

theyosh, 2020-11-27:
I guess we can close this?

theyosh, 2020-12-03:
Closing due to incidental hardware failures

closed 2020-12-03 by theyosh