theyosh/TerrariumPI #477: Need advice for temperature setting.

nukenook, 2020-11-13:
Hi,

I use DS1820 to monitor temp. and control relay board for ceramic heater. But I really confuse about the setting.

 - My day time temp. is about 30 and I want to control night temp. to 25. How can I config the sensor and environment setting?

I try to set it as

      - sensor 
                 Alarm min=25 , max=27 / Limit min=0 , max=50 / Max diff=1
      - Temperature 
                 mode=sensor / Day/night difference= -5 
      - Low alarm
                 Power on=0 / Settle timeout=120 

Is it correct?

Thank You

theyosh, 2020-11-17:
Hi,

hmm, you are close, but you should not have a Temperature difference in the environment page. As your lower limit is already 25. So with the difference of -5, it will be 20 as low limit (before the heater turns on). You should also see the alarm lines on the graph of the sensor. It should show 20 degrees as low limit during the night with the current settings.

But, what have you select for light source and light state.

I have the current setup. And I think you want the same (ignore the day/night difference)
[image]

Also, make sure you have the latest version. There was a logic error in the environment controlling.

nukenook, 2020-11-17:
Thanks for reply. So, if I change the setting to

  - sensor 
             Alarm min=25 , max=30  / Limit min=0 , max=50 / Max diff=1
  - Temperature 
             mode=sensor / **Day/night difference= 0** / **Day/night source=lights**
  - Low alarm
             Power on=0 / Settle timeout=120 / **light state=off**

Does it means that in night time (light off} if the temp. drop lower than 25 the heater will start and off after raise to 25? or off at 30?

theyosh, 2020-11-18:
The heater should kick in when it gets colder then 25 and stops when it hits 30 degrees. You should get a temperature graph that looks like:
[image]

nukenook, 2020-11-18:
OK. I got it. Thank you ^^

closed 2020-11-18 by nukenook