theyosh/TerrariumPI #479: [FEATURE]: IR Thermometer Support

cyber-shifter, 2020-11-16:
**Is your feature request related to a problem? Please describe.**
I know that use of radiant heat panels is common among people who keep reptiles. Personally I've just recently branched into them myself. They are a better way giving your reptiles the heat they need, but they come with a complication. They don't heat the air, they heat the objects instead. So, a traditional thermometer sensor won't correct detect the heat of the environment. You can make manually measurements and use an offset to SORT of get close, but this changes with the weather conditions too, while the objects are still being heated the same. 

**Describe the solution you'd like**
I'd like to see support added for some low-cost IR sensors for temperature. Once such example is this the MLX90614. It's available from different places on breakout boards, etc. This will enable an actual accurate measurement of radiant heat applied, and could be relevant to more than just reptile keepers.

theyosh, 2020-11-17:
Hmm, I guess you missed it:
[image]

But there was no Wiki. So I added that also: https://github.com/theyosh/TerrariumPI/wiki/Hardware#mlx90614

cyber-shifter, 2020-11-17:
Ah ha! Good to know! I will get some of these heading my way then. Thank you for incorporating this compatibility!

theyosh, 2020-11-27:
Is it working? Can we close this issue?

cyber-shifter, 2020-11-28:
I ordered a couple IR sensors shortly after you answered. Unfortunately nothing is arriving in a timely manner here. I have the sensor, but not the breakout board yet. I will be testing as soon as I get them (any day now, maybe even today). If you are confident that is will work properly then you can close the case. Otherwise I will once I have tested.

cyber-shifter, 2020-12-02:
Looks like it's working! Thanks for adding this, I'm sure I'm not the only citizen herpetologist that appreciates it!

closed 2020-12-02 by cyber-shifter