theyosh/TerrariumPI #481: [BUG]: random UVB and LANGUAGE problems

The-Madhouse, 2020-11-18:
Got a fresh install, i can't change thé username of password, because i have to select a language from thé drop down menu, but that menu is empty. 
Any idea's? 

Veml6075 sensor: connected correctly i guess. If i use the command to read i2c is see something connected to port 10(i guess? It's in row 0, and then line 10, and there it says 10. ) But when i try to add a sensor in terrarium pi menu and use adress ''10'' i don't get a reading.... 
What al i doing wrong?

theyosh, 2020-12-03:
Sorry for not responding earlier. But the language problem is new to me. I can only think of restarting my software, and clear your browser cache. Else we need more debug info.

And about the Veml6075. The address should be 10 . That is all. If there is no data, then there should be errors in either the screen session or in the logfiles in the log folder. https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

theyosh, 2020-12-09:
Hi, do you still have an empty language list? I cannot reproduce.

theyosh, 2020-12-16:
Is this still an issue?

theyosh, 2021-01-04:
Closing due to no updates

closed 2021-01-04 by theyosh