theyosh/TerrariumPI #482: BME280 help?

FldMker614, 2020-11-19:
Sorry if this has already been talked about, I am fairly new to this entire process so bear with me.

I have been trying for the last week to set up an automated deep water culture system for a school project that I will eventually be upscaling for my brother to use in his grow tent.

The issue that I am having is with my new bme280 sensor. It is showing up at address 76 when I look for it in bash but I am getting a null response out of it in terrariumpi. even after a reset.

here's is how I have it connected:

Physical GPIO#3: SDA
Physical GPIO#5: SCL

in terrariumpi I have it set up as a bme/bmp280 with an address of 76,1 but no dice. any advice? am I missing something or doing it wrong?

cyber-shifter, 2020-11-19:
Are you on Pin #7? I can't get a diagram in front of me.

FldMker614, 2020-11-19:
no worries. I reset the pi for a third time and it worked for some reason lol. I've got temp and humidity now. thanks tho

closed 2020-11-19 by FldMker614