theyosh/TerrariumPI #483: [BUG]: Error accessing GPIO. Make sure program is run as root with sudo!

jeryko87, 2020-11-23:
**Setup:**
 - Python version: 3
 - Raspberry PI: 4B - 4GB

**Describe the bug**
Hello, I know that a topic for this existed and was closed in the past as a hardware issue but I've just tested with two sensors and they behave the same. It looks more like a software problem.

When configuring a DHT22 sensor as per the hardware description (3.3V, GND, Physical pin 15) I get the following error in the system log:

2020-11-23 18:24:23,575 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'DHT'. Error message: Error accessing GPIO. Make sure program is run as root with sudo!

**To Reproduce / Installation steps**

The software has been installed as per the steps provided:

1. Installed Raspberry Pi OS Lite with rpi-imager
2. Set up network: set country to RO, SSIS and passphrase
3. Activated SSH
4. ssh to raspberrypi
5. sudo apt -y install git
6. git clone --recursive --depth 1 https://github.com/theyosh/TerrariumPI.git
7. cd TerrariumPI
8. sudo ./install.sh 3
9. sudo reboot
10. log in to web interface
11. add sensor as per image below. 

Group file:
adm:x:4:pi
dialout:x:20:pi
cdrom:x:24:pi
sudo:x:27:pi
audio:x:29:pi
video:x:44:pi
plugdev:x:46:pi
games:x:60:pi
users:x:100:pi
input:x:105:pi
netdev:x:109:pi
pi:x:1000:
spi:x:999:pi
i2c:x:998:pi
gpio:x:997:pi
sispmctl:x:1001:pi

[image]

[image]

Thank you

jeryko87, 2020-12-05:
Changes to BME280. Works well :).

closed 2020-12-05 by jeryko87

theyosh, 2020-12-05:
Hi sorry, but that is exactly my response... Even when using a DHT sensor....

Nice you figured it out yourself... :)