theyosh/TerrariumPI #486: [BUG]: PWM 2A AC Light Dimmer Module by KRIDA Electronics

ModestZombie, 2020-12-03:
**More info**
Searched closed tickets and found similar posts but they weren't about this specific module.

 I've been searching for a good US alternative to the Universal AC MAINS Dimmer that's documented. I've found PWM 2A AC Light Dimmer Module (linked below). Only problem is that when it's supposed to be at 0%, it's actually at 100% and vice-a-versa. So when it is supposed to be off, it is on. When it's supposed to be on, it's off. Did a test run with the Timer setting in the Environment, and that was exactly the case. Is this something that can be fixed on the software side? 

PWM 2A AC Light Dimmer Module by KRIDA Electronics - https://www.tindie.com/products/bugrovs2012/pwm-2a-ac-light-dimmer-module-50hz-60hz-120v-240v/

**Setup:**
 - Python version: 3
 - Raspberry PI: 3B+

**Describe the bug**
When on the Switch Status page, and it's supposed to be at 0% (OFF), it's actually at 100% (ON) and vice-a-versa
So when I tell the dimmer to be at 75%, in reality, it's turning on to 25% power. And if I tell it to down to 25%, it will actually get brighter and appear to be at 75% power. Whether I adjust it here, or set up the timer - it does the opposite of what I want it to. 

**To Reproduce / Installation steps**
See the screenshots of how the switch is setup with the various fields and their values. Used the switch profile 'PWM Dimmer'. I've followed the instructions from Stewarts blog but modified the values just a wee bit. 
[image]
[image]

**Expected behavior**
I expect it it to turn off when it's at 0%, essentially I'd like it to function as it's intended. 

**Screenshots**
[image]
I copied and pasted the log in a txt file.
[Syslog.txt](https://github.com/theyosh/TerrariumPI/files/5634167/Syslog.txt)

**Additional context**
The seller of the PWM Dimmer provides this as an example if your were to run it though an Arduino, don't know if this would helps but:  https://drive.google.com/file/d/0B6GJokXFb5oEdFAxVEItdWN2S1U/view?usp=sharing

As a side note, originally ordered this version (v2) from Amazon (https://www.amazon.com/gp/product/B06Y1DT1WP/ref=ppx_yo_dt_b_asin_title_o00_s02?ie=UTF8&psc=1), but it was either faulty or defective because it would automatically turn on at full blast when plugged in.

theyosh, 2020-12-03:
Hmm, could it be, that by adding it as a IRF520 Mosfet dimmer will work? That dimmer is working in the opposite (inverse) way of a PWM dimmer. I am not that familiar with PWM dimmer. I guess in a few days we get a nice explanation of @AndrejWeber . He knows a lot of PWM dimming.

ModestZombie, 2020-12-03:
I'm in the same boat as you, I'm not too familiar with PWM dimming but your guess would be much better than mine! I'm okay waiting to hear if Andrej has any more input on this. Thanks for taking the time to look into this!

theyosh, 2020-12-03:
I would at least try the Mosftet type. It cannot break much. Just do not touch it by hand when you are testing ;)

ModestZombie, 2020-12-03:
Wow, I totally misread your last comment -- YES, adding it as the IRF520 Mosfet dimmer solved the issue! However, now a slightly separate issue, it seems to flicker way more when added as a Mosfet dimmer. I've added and removed it a few different times to  confirm. I keep the same exact settings, except changing the type of dimmer added. I can upload a video of the light bulb in action if need be, so you can get a better idea as to what I'm talking about. Any suggestions? Thanks again for your time.

Setting it as a PWM Dimmer: Seems to increase in more obvious, steady increments. Still holds steady (with an occasional flicker) at 50%. Sometimes moving it up or down a few percentages and the flicker stops almost entirely. 
Setting it as a Mosfet Dimmer: Flickers on pretty much any percentage, but even at 50% it's pretty bad. Past 80% power it doesn't flick _as much_ but it can still be noticeable until you crank it up to 100%.

AndrejWeber, 2020-12-03:
I will have a look later this evening. 
I can imagine, that the frequency of PWM signal is not right. May be it should be higher or lower. At quick look I have not found any documentation for this dimmer.

theyosh, 2020-12-03:
@ModestZombie if you are a bit handy, you can play with the following lines of code:

https://github.com/theyosh/TerrariumPI/blob/master/terrariumSwitch.py#L1188

Put the value of DIMMER_FREQ to 500 and rerun the sofware. And see if this will reduce the flickering.

@AndrejWeber no worries, no rush. I think it is a frequency value, so just test some values

theyosh, 2020-12-03:
Hmm, in my 4.X version, I use the frequency of 5500. But that code is different. But give it a try

ModestZombie, 2020-12-03:
@theyosh, that made it significantly better but its not _quite_ as good as the PWM Dimmer profile. @AndrejWeber what's hard is the documentation on this module is close to non-existent. The seller has an eBay, Amazon, and Tindie but they all contain the same basic (and non-helpful) information on it. He does give an example of the code he uses with an Arduino that I linked in the original post, but I don't know if that will provide any insight. It does support "PWM input signal up to 10kHz" - I don't know much about this though, am I safe to try increasing the value of DIMMER_FREQ to 5000? Or does the value of DIMMER_MAXDIM on the line right above it need to be adjusted as well?

Thanks again for both of your suggestions and time.

AndrejWeber, 2020-12-03:
You can also try 10.000
That should be the maxim, the dimmer can handle.

AndrejWeber, 2020-12-03:
You also can edit line 1179 at DC-dimmer to 10000
The dc-dimmer is quite the same like pwm-Dimmer, but it has inverted signal. So you will get 100% on at 100% Dimmer-state

ModestZombie, 2020-12-03:
Gentlemen, **thank you both**. 
As of this point, I've modified the DIMMER_FREQ on both lines 1179 and 1188 to 10000 and after testing out both options, they are performing remarkably well! Initially, it seems the DC Dimmer is the winner but barely, they both have almost no flicker at all except for right after it's done making an adjustment to the power (which is more noticeable on the Mosfet setting, but _barely_). I can totally live with this (and I'm sure my Boa can too!). Now to finish setting up the rest of the sensors!

I'm going to go ahead and order a few more of these dimmers from his Tindie store. I am also (attempting to) documenting my process, and will put together a tutorial in case others would like to use this same module. It may take me until just after the holidays to complete it though. 

Follow up question that you may or may not be able to answer: Would there be a reason why I would want to have the DIMMER_FREQ at 5000 instead of 10000, if it has the same reasonable amount of flickering? Like does having a higher frequency setting equate to higher CPU usage or anything like that?

AndrejWeber, 2020-12-03:
There is no reason to use 5k instead of 10k. 
The both pins on raspi use hardware PWM so there is almost the same cpu usage by using different frequencies. 
All other gpio can only do software PWM, so there you will increase the cpu usage by rise the frequency. 
Of course, if you will use a second AC-dimmer, which provides a maximum frequency of 5k or some else, then you have to use the lower one or you can use a different dimmer type like dc-dimmer. Then you can use two different frequencies. 
You should know, a PWM signal is just a very fast switch, that will turn on and off the 3.3V 100 times a second at 100Hz setting. 
For example: at a low dimmer setting like 1% the the "switch" will turn on only 1/100 of a second and turn off for 99/100 of a second. 
By increase the frequency you will have much more "switch on" every second. At 10kHz you have 100 "on"-signals every second. That will look much more smooth. 
At last, your dimmer will also work with higher frequencies than 10k theoretically up to 8MHz. But somewhere at higher frequencies, there will be a effect, where the cable and the dimmer layout will smooth the signal through capacitive effects. Then the chip cannot register any PWM signal and will stop dimming or just will start flickering.

theyosh, 2020-12-04:
Yeah, so basically you need a calibration value for the frequency. Also I noticed some frequencies will produce a high pitch when I was working with cooler fans. Strange...

But that also means that I have to have two PWM dimmers. And that is it. And in the new version, it looks like I will end up with that. Because I have one with active up pull / inverse and all the normal and DC versions work the same as I look to the code.
Only the frequencies could differ. So, maybe I should add that for a dimmer as a calibration option.

ModestZombie, 2020-12-08:
Now that I've had another moment to test this out again, and doing some additional reading on top of your comments (great explanations btw) - I've found the sweet spot for the DIMMER_FREQ for now: 2.8k. There is _little_ to almost no flickering. Putting my dimmer at the max 10k made it do pretty much like you explained, @AndrejWeber. It would technically still dim decently at 10k, but it would flicker in an odd pattern.

@theyosh I think I understand what you mean. After my own testing and switching back and forth between different dimmer profiles, I could see how this would benefit from the changes you mentioned. Especially since they all work the same, as long as you give the user a Frequency Field, a Normal/Inverted Polarity option (unless that will be the difference in the two PWM dimmers you mentioned), and I'm assuming a Max Dimming Value - you wouldn't have to create a profile for every dimmer out there.

theyosh, 2020-12-18:
Ok, then I would like to ask @ModestZombie if it is workable to manually patch your version for now? In the new version, I have made the dimmer frequency an input value. In order to make it for the current version it pretty hard at the moment.

[image]

Because then I can close this issue. And in the new version, it should work better and easier.

cyber-shifter, 2020-12-21:
I've read through this post. In my particular case, I'm using radiant heat panels instead of lights. Could I use this to dim the radiant heat panel power as well. Usually radiant heat panels are controlled by varying the power to them, similar to dimming a light. I would rather control them in way that keeps them at a consistent power level than constantly turn them on and off like with a relay.

theyosh, 2020-12-21:
@cyber-shifter that is a different issue. Please just try it. I do not have the answer for this.

cyber-shifter, 2020-12-21:
That's cool. I figured I would ask. I've not messed around with PWM dimmers much either.

ModestZombie, 2020-12-22:
@theyosh I ran into an issue attempting to manually update (I followed the instructions on the main page for updating, but I didn't stop the TerrariumPi service from running before I updated), so I formatted my SD card and installed a fresh version of TerrariumPI. **1)** Am I supposed to do anything different to get the updated version? It's still the old layout/options for me. Tried accessing it in a Incognito window as well just to make sure it wasn't the browsers cache. 

Full disclosure, as I was testing the PWM Module by KRIDA - I broke it somehow. Hard to say if it was my fault or the lamp/dimmer. I had been testing various frequencies for DIMMER_FREQ because the previous value was still flickering erratically, and had also changed the DIMMER_MAXDIM a few times. Heard a loud pop from the lamp, then immediately afterwards a pop from the dimmer - followed by a trail of smoke. The dimmer no longer dims and just is always on. The lightbulb still works fine though, but I haven't dared to pair another dimmer with that particular lamp. Two more of these dimmers just arrived in the mail, however due to safety concerns, I also ordered the MDPM4.1 Dimmer (as well as the 7.4 version) which arrived last Friday (I _really_ should have ordered these from the very start). I like the idea of it just blowing a fuse, rather than potentially catching my snakes cage on fire should something happen like that again.

Maybe I'm not understanding something fundamental, but I have a follow-up question to either theyosh or @AndrejWeber, if you happen to know. **2)** What is the correlation between DIMMER_FREQ and DIMMER_MAXDIM? How is that used to create the Duty Cycle? How did you come up with those values that are used in the equation with the other dimmers, as shown here: https://github.com/theyosh/TerrariumPI/blob/b678be1d881f7bab93cfc0dbf085fdb2f6f416fb/terrariumSwitch.py#L1101
I'm trying to eliminate as much guess work as possible, but I understand I'm treading new waters with this so I'm willing to give anything a shot. 

Last question,  **3)** regarding the MDPM4.1 dimmer, it's working AMAZINGLY well, but my bulb seems to have reached max brightness around 55%. Can this be modified simply by changing the DIMMER_MAXDIM value? Thanks again for your time and knowledge.

cyber-shifter, 2020-12-22:
@ModestZombie I JUST order two of those KRIDA dimmers to mess with using them to control my radiant heat panels. Are they not suitable?

AndrejWeber, 2020-12-22:
@ModestZombie 
The frequency is the amount of (on/off) signals every seconds. So if you will set 20% and 1Hz frequency, the signal will be on for 20% of a second, so just 0.2 seconds on and 0.8 seconds off. 
That is much to slow for the hardware we use with PWM.
If you will set to 100Hz, you will do the 0.002s on and 0.008s off. So it will repeat 100 times per one second. The max dim value is the real max value, like you have told about 50% with full brightness. 
So the dc-dimmer you can set in the gui will work from 0% to 100% PWM, but the real PWM signal is only 0-50%. 
The PWM-dimmer will set 50-100%, how I can remember. 
It is just mathematics to have a better usability. You have to test, what will be your best max value.

theyosh, 2020-12-22:
@ModestZombie Great job! Frying hardware is bonus points for me! That is why I always buy in pairs or more of the same device/sensor. But to bad that dimmer is gone.
@cyber-shifter I do not think that it is a big issue. During testing this can happen. Also you want to use heating panels, which do not provide light. So the 'flickering' is not a big issue when not using lights.

@ModestZombie About the current dim values: I started it I think about 3 years ago with the dimmers. I did not have any knowledge and by trial and error I figured out those values for the MDPM4.1 dimmer. And based on the code of the maker.
And from there, I made the other dimmers, but all without testing (much). So I assumed it was all ok. 
But in the new version, I only have to set the dim frequency and the dim value from 0 - 100. And that looks good when I test it with several dimmers. But still I cannot explain it more then @AndrejWeber does. I am also playing with dim-able led lights, but they are also not all the same... So, for that, I leave it up to the person that is using it, to figure out the correct values and hardware. I cannot test all kind of hardware. To expensive and I do not have the room for that.

Then answereing question 3. You can use the field  'Dimmer on percentage' . Enter the max value, in your case 55, and the dimmer will stop at that amount. So it limits the max power that can go through. And the switch should not be put above 55. The gui has a bug, where the change looks ok, but when you reload the page, the old value is back.

And another thing with dimmers is that the dim value sometimes means let that amount power through vs. the amount is the actual dim amount, and will let through 100 - value. So sometimes is it the actual dim value, and sometimes it is the value to let trough. And that can cause some strange issues when not taken care off in code.

theyosh, 2020-12-22:
To get the latest version, just use `git pull` on the master branch. That will always get you the latest version.
Also, updating when running is not that big of an issue if the version numbers are the same. I do a lot of updates without releases. Bad practise, but I hate making releases. Then also handy: https://github.com/theyosh/TerrariumPI/wiki/FAQ#which-files-to-backup

ModestZombie, 2020-12-22:
@cyber-shifter 100% what theyosh said. for your radiant heat panels, they should be fine. It definitely works as a dimmer, as for controlling lights though I couldn't get it to stay steady (or rather, not flicker erratically). So far I haven't gotten it as smooth as MDPM4.1 can. Thats not going to matter in your heat panels, and you probably won't even notice it. If I find a good frequency value for the KRIDA dimmer, I'll make sure to share. 

@AndrejWeber Thanks for the explanation, looks like I'm a master of confusing myself, I appreciate the explanation. I also think I was misinterpreting the code. Yosh's detail about how sometimes the "dim value sometimes means let that amount power through vs the actual dim amount" and it made me go re-examine the code. More testing is needed on my end, for sure. and speaking of - @theyosh I'm still not able to get the newer version with the Freq field. What I've done so far - 
[image]
After this, I ran `sudo reboot`
Now, running `sudo git pull` again and it says "Already up to date."
I've removed and readded the Mosfet dimmer but I'm not seeing those updated fields:
[image]

So do I still need to run `sudo ./install.sh 3` because last time, after reinstalling - thats where I ran into some errors and TerrariumPi was no longer reachable after a reboot. I've made a backup image of my sd card in the previous build so if just need to try that again I can. but any suggestions would be appreciated. Having that Freq field would make thing _much_ easier for testing lol which I'm willing to do since have two more of the KRIDA dimmers, and now 2 of the MDPM4.1 and 2 MDPM7.5.
 
Thanks again for everyone's time!

theyosh, 2020-12-23:
@ModestZombie Sorry, the update I referred to is not live. I am working on a new version, which is a complete rewrite of the code.

There I changed the dimming code a lot. And easier to setup. But that is not available for a couple of months. That is why I asked if you are able to patch your own version for the right dimming values.

ModestZombie, 2020-12-27:
@theyosh That for clarifying, my apologies on the misunderstanding but that makes sense. Thanks for all your time and work on this. Upon testing, I was able to find a frequency that has worked quite well for the time being for the KRIDA dimmer. And so far, nothing has started smoking LOL

I used the RF520 Mosfet dimmer profile, and changed the DIMMER_FREQ [on line 1190](https://github.com/theyosh/TerrariumPI/blob/eb24c75aa2e63f8ef32bb047e1fa618cb9c40088/terrariumSwitch.py#L1190) to at least 240. 300 seems to work well. 

**Steps I took:** 
SSH into your Pi and run the following
`cd TerrariumPI`
`nano terrariumSwitch.py`
Then press **Ctrl** + **-** at the same time. Enter 1190, which should take you to the line we need to edit.
Change the DIMMER_FREQ to at least 240  (300Hz also works, even a little bit better. Previously, I was trying values that were too high, like in the 1KHz - 10KHz range. When the freq is too high, you get some erratic flickering due to the signal being out-of-sync with the AC. From what I've read, it sounds like you may be able to fine tune this frequency value if you have an oscillator, which I do not have.)
Leave DIMMER_MAXDIM = 100 (Previously, I was changing this value while also trying different DIMMER_FREQ which was messing me up further. Don't change it unless you need to)
Hit **Ctrl** + **x**, then hit **y** followed by **Enter** to exit and save changes. 

@cyber-shifter in case you haven't found a good setting for that dimmer, try those above. Maybe if i have a little more time I will test out other values and see at what frequency it starts to get too high at, but for now this should work.

Thanks again for the help and information @AndrejWeber and @theyosh! This particular issue has been resolved.

closed 2020-12-27 by ModestZombie

cyber-shifter, 2021-01-20:
@ModestZombie so I got some of those KRIDA dimmers. I set them up as DC Dimmers like was mentioned above, since it seemed to work better for you. However, even though the it will go up to 100, the power never seems to get all there. Is it because on DC-Dimmer the PWM only goes to 50 percent? I need the panel to get higher than 50, but not all the way to 100. I looked at the post right above this one, so I'm going to try those setting, but which hardware setting are you using with that?

ModestZombie, 2021-01-20:
Instead of the DC Dimmer profile, try the RF520 Mosfet dimmer profile. That's what I'm using for the time being (with a Freq of 240. I'm not really sure how the Freq value affects or benefit a RHP, other than maybe a more stable/consistent power draw). 
If that still didn't work, You could try the PWM Dimmer profile. Just keep in mind it will have reverse polarity, so 100% = off, and 1% = On. If you leave your panel at 1% on the PWM Dimmer profile and it still only seems to get to 50% power, its possible you may need to adjust the DIMMER_MAXDIM value on the DC Dimmer or Mosfet Dimmer profile but I'm not sure what to, or if it's an issue with something else entirely. I would open up a new ticket about your issue specifically. I can try to help you more there, that way Yosh and other folks can help too.

cyber-shifter, 2021-01-20:
@ModestZombie yeah, I'm going to open a new issue if this doesn't work. I just wanted to clarify on which profile you were using. I'll give this a shot, and if I still have issues, I'll definitely open a case.

threedom, 2021-03-04:
@ModestZombie I am also using a Krida dimmer [http://www.inmojo.com/store/krida-electronics/item/2-channel-ac-led-bulb-dimmer-module-v2/]

No matter what I do with the frequency, both LED and incandescent bulbs flicker. Any advice / ideas as to what might be the issue? Some additional context of what I tried and what results I got:

When I change the frequency setting the following behaviors are triggered:

<110 flickering
at 119 smoothly drops from max brighness to almost 0 in about 1 second (in a loop)
at 120 brightness increases smoothly from off to 100% and then abruptly switches off (speed of this process depends on what percentage is selected in TerrariumPI)
120 (e.g. 140) flickering

at 240 brightness increases smoothly from low to 100% and then drops back to low
very high (e.g. 720, 24k) only switches on to full brightness (sometimes only at high percentage in TerrariumPi
Long story short:
It looks like somehow the dimmer should be able to just maintain a lower brightness since it can smoothly transition the bulb through brightness levels. But somehow it doesn't maintain one brightness level.

Thanks so much for any attempts to help!