theyosh/TerrariumPI #487: At a complete loss

cyber-shifter, 2020-12-04:
It's going to be something dumb, I can feel it. So, I got my setup working, everything looks good. Except the Pi had a funky IP address from the DHCP server, so I wanted to change it. I had previously setup a hostname for the Pi, and told TerrariumPi to look for the hostname. That all worked. So, rather than doing a static lease in my router (because I wanted the device to maintain it's IP if the router failed for DR purposes), I just edited the /etc/dhcpcd.conf file, following the guidelines that are actually in the file. Once I made the changes, and rebooted, Terrarium Pi stopped working. I can SSH to the raspberry pi on the IP address. If I ask it for the hostname, it gives me what it is supposed to be. I can even ping the device. But when I got to either the hostname or the IP address I get a connection refused. The server is running, I did a git pull to the latest, even though I already had the latest. I'm at a lose for what to check here. I don't understand why I can reach it, but I can't reach it.

cyber-shifter, 2020-12-04:
Completely forgot a detail. I commented the lines back that I changed in the config file, and instead tried a static lease in the router. No dice.

cyber-shifter, 2020-12-04:
I looked at this again this morning with a clear head. It must be something in the router, because the other Raspberry PI that I have is broken too, and I didn't touch it. Fun.

cyber-shifter, 2020-12-04:
This has been a rollercoaster... Both raspberry pi's are on 3.9.8. Previously they weren't, and one would still work, but I upgraded the other one as part of my troubleshooting the static IP thing.

cyber-shifter, 2020-12-04:
Checked the settings.cfg file, I had previously set the hostname to the hostname of the machine. I guess it didn't like having just the hostname there? Do I need to append it with .local or something? Once I set hostname back to all it worked again. I'm going to try setting the static IP again later today.

theyosh, 2020-12-04:
Put the hostname always to 0.0.0.0 . This will make it listen on all IPs.

If you fill in a hostname, then the software has to ask your router dns for the IP. And if there is a network issue, and that the entered hostname is not resolvable, it will not start.

cyber-shifter, 2020-12-08:
I got it working again, by defaulting it to the :: option that the tooltip says, but I'll try the 0.0.0.0. I recently bought a new router and forgot I don't have my own DNS server internally anymore yet. So that was part of it. I'll go ahead and close the issue.

closed 2020-12-08 by cyber-shifter