theyosh/TerrariumPI #488: [FEATURE]: Lines on graphs to show when switches are turned on/off

cnelsonsic, 2020-12-04:
**Is your feature request related to a problem? Please describe.**
When I look at the dashboard, I see the current state of the environment, and some graphs which give me some historical information. If the temperature graph goes down, I'd like to know if it's because my heater turned off or because I left the door open, for example. I would also like to know how much lag time (heater has to warm up) between turning it on and seeing an effect on the environment, and also the hang time (heater cools off, temperature stops rising) after turning it off. This would help me fine tune my timing for my devices and prevent inefficient power cycles.

**Describe the solution you'd like**
If I were to see a vertical line on the graph, and be able to hover over it to see "Heater: On @ $timestamp" or something similar, I'd be able to see when my heater turned on. This would help me visualize the effect of my devices on the environment.

**Additional context**
I may be imagining it, but I feel like there's something like this already in place for door switches, so maybe that would be reusable across all the graphs?

cnelsonsic, 2020-12-04:
I see what you mean, that's a much more involved feature than I realized. It's definitely more of a wishlist item.

theyosh, 2020-12-05:
But it is what you where asking for right? In the new system, I think I can make this more easily, as I can now already take some measures for it.

cnelsonsic, 2020-12-06:
Honestly all I'm looking for is vertical lines when a switch changes state. I'm imagining something that looks like this: https://www.dummies.com/wp-content/uploads/425738.image0.jpg

It's not so useful for dimmers, but it might be something to keep in mind if you touch the graphing parts.

theyosh, 2020-12-09:
Hmm, for that, maybe if you are handy, you could create your own data insights by using the export function of the graphs. It produces CSV files with the data. That give more freedom to make your own graphs of alarms.

theyosh, 2022-06-26:
Hi @cnelsonsic 

I am afraid that I will not implement this. As it is hard to make, and I am still not sure how to combine sensor graphs and relay actions.

Sorry for this... but I cannot implement everything.

closed 2022-06-26 by theyosh

cnelsonsic, 2022-06-27:
No problem, you were messing with the graphs at the time and I figured it'd be a good time to mention the idea, if it was even possible.

This would really be in the realm of visualizing data outside of TerrariumPI, to be honest.

I may come back to this at some point, if I do I'll write up a little post about what I did.