theyosh/TerrariumPI #490: MOSFET for LED control

cyber-shifter, 2020-12-08:
If you have this already and I missed it, I apologize. I ran out of time this morning to look more thoroughly. I am thinking about putting some LED strips into my enclosures, that will dim with sunrise and sunset, just to emulate the light cycles of the day. My new enclosures are different, and not glass. So I wanted to see if I could use a MOSFET to control an LED strip, or if there was another way you had already to do it. If not, I could always dim the mains to the adapters that power the LED's, but that seems inelegant.

theyosh, 2020-12-16:
Sorry, I do not know. I would just try and see if it works. I cannot support all kind of dimming and led questions...

Also, leds, can be dimmed by changing the brightness. Which is not reducing power to it, but telling to use less power. Which is different programming.

cyber-shifter, 2020-12-16:
Cool. I'll look into it further. I just wanted to ask in case you already had something, to try to stick to standard config.

closed 2020-12-21 by cyber-shifter