theyosh/TerrariumPI #493: [BUG]: Storage usage shows warning but enoug free space available

Dave0408, 2020-12-18:
**Setup:**
 - Python version: 3.7.3
 - Raspberry PI:  Pi Model B+  V1.2
 - SW Version: 3.9.8

**Describe the bug**
Incorrectly, the memory usage is displayed as a warning. However, there is still plenty of physical memory available.

**Expected behavior**
Display should indicate that everything is OK here. 

**Screenshots**
```
pi@raspberrypi:/ $ df -h
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
/dev/root        59G    5,1G   51G   10% /
devtmpfs        153M       0  153M    0% /dev
tmpfs           185M       0  185M    0% /dev/shm
tmpfs           185M    7,4M  178M    4% /run
tmpfs           5,0M    4,0K  5,0M    1% /run/lock
tmpfs           185M       0  185M    0% /sys/fs/cgroup
/dev/mmcblk0p1  253M     54M  199M   22% /boot
tmpfs            37M    1,6M   36M    5% /run/user/1000
```

[image]

theyosh, 2020-12-21:
Hm, which graph is this? memory or disk?

And the graph shows a value of 10  times the actual value? Because I a line close to the max value.

What is the outcome of: http://[TERRARIUMPI_PI]:8090/api/system ?

And since when is this? You can see the graph values here (big file): http://[TERRARIUMPI_PI]:8090/api/history/system/disk

I can't find anything that is causing this. So or it is something specific to your setup, or a really strange case.

Dave0408, 2020-12-22:
The warning is about disk space. Following is the output of the two commands. I can't find anything really suspicious here. The original file is in the attachment. 

http://[TERRARIUMPI_PI]:8090/api/system:
```
{

   "memory":{

      "total":387174400,

      "used":177119232,

      "free":210055168

   },

   "disk":{

      "total":62555754496,

      "used":5521645568,

      "free":54442934272

   },

   "load":{

      "load1":0.4,

      "load5":0.34,

      "load15":0.29

   },

   "uptime":250699.81,

   "cores":1,

   "temperature":35.78,

   "external_calendar_url":""

}
```
http://[TERRARIUMPI_PI]:8090/api/history/system/disk
```
{
   "system":{
      "disk":{
         "total":[
            [
               1608526500000,
               62555754496
            ],
            [
               1608526560000,
               62555754496
            ],
            [
               1608526620000,
               62555754496
            ],
            [
               ...,
               ...
            ],
            [
               1608612720000,
               62555754496
            ],
            [
               1608612780000,
               62555754496
            ],
            [
               1608612840000,
               62555754496
            ]
         ],
         "used":[
            [
               1608526500000,
               5518163968
            ],
            [
               1608526560000,
               5518163968
            ],
            [
               1608526620000,
               5518168064
            ],
            [
               ...,
               ...
            ],
            [
               1608612720000,
               5521645568
            ],
            [
               1608612780000,
               5521653760
            ],
            [
               1608612840000,
               5521657856
            ]
         ],
         "free":[
            [
               1608526500000,
               54446415872
            ],
            [
               1608526560000,
               54446415872
            ],
            [
               1608526620000,
               54446411776
            ],
            [
               ...,
               ...
            ],
            [
               1608612720000,
               54442934272
            ],
            [
               1608612780000,
               54442926080
            ],
            [
               1608612840000,
               54442921984
            ]
         ]
      }
   }
}```

[data.json.txt](https://github.com/theyosh/TerrariumPI/files/5727878/data.json.txt)

theyosh, 2020-12-22:
Found the issue. The alarm badge would also toggle on when there is less then 10% of disk storage. In your case the limit is 6,4GB. And 5.1GB is less and the alarm is triggered.

Changed the low alarm just to zero.

Do a `git pull` and restart. It should be fixed now.

Dave0408, 2020-12-23:
Thanks. Works now as expected.

closed 2020-12-23 by Dave0408